danielcherubini/elegoo-homeassistant #12: Camera

danielcherubini, 2025-02-07:
### Checklist

- [x] I have filled out the template to the best of my ability.
- [x] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [x] This issue is not a duplicate feature request of [previous feature requests](https://github.com/ludeeus/elegoo_printer/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

No

### Describe the solution you'd like

I would like to be able to view the camera

### Describe alternatives you've considered

Tried creating a RTSP stream through HA and it didnt' work.

### Additional context

Seems that when creating the stream, i get the error in logs

``` [homeassistant.components.stream.stream.camera.saturn_4_ultra_camera_image] Error from stream worker: Error opening stream (INVALIDDATA, Invalid data found when processing input, rtsp://xxxxxx:554/video)```

Looking at similar errors, it would appear that something is a little weird with both how the elegoo camera sends RTSP video streams and also limitations in HA.

This might not be able to be done.

danielcherubini, 2025-02-07:
We might not be able to do this.... we might need ffmpeg to transcode

https://github.com/AlexxIT/go2rtc/blob/master/internal/ffmpeg/ffmpeg.go

https://www.reddit.com/r/ElegooSaturn/comments/1i0qv6a/comment/m78v69p/?context=3

fynflood, 2025-03-11:
I have the camera working using the Generic Camera integration. It's worth noting that after hitting submit on the settings the next dialog box gets kind of jittery, but hit "it looks good" and continue - seems to work fine for me. 

[image]

danielcherubini, 2025-03-11:
2 things..
1. I see you didn't add the `/video` at the end
2. 2hz.. let me check this

danielcherubini, 2025-03-11:
OK I've got it working 100% of the time using the go2rtc plugin!

```yaml
streams:
  test: ffmpeg:rtsp://{{YOUR_PRINTER_IP}}:554/video?#input=rtsp/udp#video=h264#media=video#resolution=960x540
```

danielcherubini, 2025-03-11:
This uses the following flags...
```"source": "exec:ffmpeg -hide_banner -v error -allowed_media_types video -fflags nobuffer -flags low_delay -timeout 5000000 -user_agent go2rtc/ffmpeg -i rtsp://{{IP_OF_CAMERA}}:554/video? -c:v copy -an -user_agent ffmpeg/go2rtc -rtsp_transport tcp -f rtsp rtsp://127.0.0.1:8554/fe934b6ac6a9ce3711fce69fd1fbb734",```

danielcherubini, 2025-06-06:
Apparently ... for the Centuari Carbon, the printer is MJPEG and a different url

Thanks to @Polinias

> > Video, yes this will support that. The issue is that the rtsp stream isn't standard, and needs processing which can be CPU intensive. There's a PR open.
> 
> Great addon, thank you!
> 
> Regarding camera feed. It is working out of box using MJPEG IP camera integration, so no need to make any magic stuff. Stream is available at http://IP-ADDRESS:3031/video.
> 
> Looking forward to fixed times and ability to control printer, through.

danielcherubini, 2025-06-29:
Released in v0.14.0

PR #197 (2025-08-16): Feature/ffmpeg camera 2
Fixes #12 and #196 

## Summary by CodeRabbit

- **New Features**
  - FFmpeg-based camera support for resin printers: low-latency MJPEG streaming, still-image snapshots, and a new camera entity.
  - New per-printer options during setup: enable camera for resin printers; enable proxy for FDM printers.

- **Bug Fixes**
  - More robust stream URL handling, availability checks, and proxy behavior.
  - Printer model matching is now case-insensitive.

- **Chores**
  - Added ffmpeg dependency, bumped package version to 2.1.0b5, updated .gitignore.

- **Tests**
  - Updated unit [... cut]
closed 2025-06-29 by danielcherubini