danielcherubini/elegoo-homeassistant #21: Vat Temp

fynflood, 2025-02-13:
### Checklist

- [x] I have filled out the template to the best of my ability.
- [x] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [x] This issue is not a duplicate feature request of [previous feature requests](https://github.com/ludeeus/elegoo_printer/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

Would love to see the vat temp available, and if a preheat function exists (I can't do it from Chitubox, so maybe not?) that would be great. 

### Describe the solution you'd like

an entity for vat temp

### Describe alternatives you've considered

I haven't considered any

### Additional context

If it were available, I could automate the ventilation system to only kick on when the tank is up to temp since having it on in the winter hinders it some.

danielcherubini, 2025-02-13:
Thanks. Yeah I can do this super easy. I'll work on it in the next day or so.

fynflood, 2025-02-13:
Hm - using a Saturn 4 Ultra 16k with the latest firmware here, the only temp entity available is the UV LED Temp

danielcherubini, 2025-02-13:
You mean in this integration? Yeah that's because it's the only one i exposed.. i've got the regular Saturn 4 Ultra, so i never exposed it fully.. just internally.

I'll have some time tomorrow.

danielcherubini, 2025-02-13:
I had some time tonight :D

danielcherubini, 2025-02-13:
There's no way i can test these.. I'll leave these as "Box" temps for now.. but i believe these are the vat temps.

Also, they will say "Unavailable" if i get no data from the printer... so just let me know here if it works or not.

danielcherubini, 2025-02-13:
You can force the update to check for the new version

[image]

fynflood, 2025-02-14:
Updated! Doesn't seem that box temp is the way. I thought maybe it needed to be printing for it to update, but no dice.

[image]

danielcherubini, 2025-02-14:
Yeah so until chitu-systems updates the payload in SDCP, then I guess we're out of luck. 

Shame, that would have been cool!

fynflood, 2025-02-14:
Thanks for the effort anyway!

danielcherubini, 2025-02-14:
Anytime! 

current priority is getting the full status object into HA

https://github.com/cbd-tech/SDCP-Smart-Device-Control-Protocol-V3.0.0/blob/main/SDCP(Smart%20Device%20Control%20Protocol)_V3.0.0_EN.md#status-information

Is there anything in here you think might be something you want?

fynflood, 2025-02-14:
Timelapse status would be handy, nice to know if it's on or not.

Thumbnail would be legit!
https://github.com/cbd-tech/SDCP-Smart-Device-Control-Protocol-V3.0.0/blob/main/SDCP(Smart%20Device%20Control%20Protocol)_V3.0.0_EN.md#retrieve-task-detailscmd-321

danielcherubini, 2025-02-14:
I'm actually going to work on that thumbnail next actually...  mostly to improve my notifications

danielcherubini, 2025-06-06:
@fynflood ... do you mind doing something?

Can you pull this repo, run `make` 

then edit debug.py, put in your ip_address, make sure centuari_carbon is false, then run `make debug`

Paste in the status object here.. I have a feeling that you might have extra data in that object, that isn't in the spec.

danielcherubini, 2025-06-21:
@bushvin you have a 16k right? If so, can you use `make debug` and check the status or attributes objects to see if there's any new data?

danielcherubini, 2025-06-21:
This should be done now in v0.10.0, please remove the printer and re-add it in the integration.

closed 2025-06-21 by danielcherubini

danielcherubini, 2026-01-07:
Thanks. It's temp of tank. 

I'll get that and the target into the next version.

koshisan, 2026-01-07:
Wonderful! As far as I can see its:

"HeatStatus": 0, <- 0 = Heater is off, 1 = Heater is on 
"TempOfTank": 15, <- Current temperature
"TempTargetTank": 30, <- Target temperature of tank - like Release Film Max this is fixed in Firmware and cannot be adjusted

Of course it would be great if there was an option to toggle the pre-heating remotely, but it looks like this isn't even exposed through the API...

danielcherubini, 2026-01-08:
Fixed and tested in https://github.com/danielcherubini/elegoo-homeassistant/releases/tag/v2.5.1