danielcherubini/elegoo-homeassistant #28: Unknown error when adding a printer in another VLAN

thiesch, 2025-03-03:
### System Health details

## System Information

version | core-2025.2.5
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.13.1
os_name | Linux
os_version | 6.6.73-haos
arch | x86_64
timezone | Europe/Berlin
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
HACS Data | ok
GitHub API Calls Remaining | 4980
Installed Version | 2.0.5
Stage | running
Available Repositories | 1564
Downloaded Repositories | 16

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 14.2
-- | --
update_channel | stable
supervisor_version | supervisor-2025.02.4
agent_version | 1.6.0
docker_version | 27.2.0
disk_total | 30.8 GB
disk_used | 17.3 GB
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
virtualization | kvm
board | ova
supervisor_api | ok
version_api | ok
installed_addons | File editor (5.8.0), MariaDB (2.7.2), InfluxDB (5.0.2), Grafana (10.4.2), ESPHome Device Builder (2025.2.1), Filebrowser (2.23.0_14), Studio Code Server (5.18.3), Matter Server (7.0.0), Terminal & SSH (9.16.0), Mosquitto broker (6.5.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 3
[... 110 lines cut ...]
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/data_validator.py", line 74, in wrapper
    return await method(view, request, data, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/data_entry_flow.py", line 121, in post
    result = await self._flow_mgr.async_configure(flow_id, data)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 340, in async_configure
    result = await self._async_configure(flow_id, user_input)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 387, in _async_configure
    result = await self._async_handle_step(
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        flow, cur_step["step_id"], user_input
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 543, in _async_handle_step
    result = await self.async_finish_flow(flow, result.copy())
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 1500, in async_finish_flow
    title=result["title"],
          ~~~~~~^^^^^^^^^
KeyError: 'title'
```

danielcherubini, 2025-03-03:
What's interesting is none of that stack trace is from this integration. A lot of it is roborock and the actual stack comes from home assistant core.

thiesch, 2025-03-03:
I think I have found a clue.
The last exception of 06:47:35:882 can be reproduced every time the IP is added. 

The problem seems to have less to do with the VLAN than that the IP of my NGINX server is addressed in the error handling.

danielcherubini, 2025-03-03:
Thats interesting... so, how are you adding the printer? by IP? or are you using nginx to do dns reverse proxies?

danielcherubini, 2025-03-03:
also.. just to test it.. try putting the printer in the same vlan, and check if it has the issue. If it does then i expect the issue might not be this plugin.. but something else, as the ip that aiohttp was having issues with was your nginx server (which is real odd)

if it doesn't then we can continue.

thiesch, 2025-03-03:
I try to add the printer directly via the IP address, which only results in the exception. 
I use nginx purely for external access to the Homeassistant instance. 
The problem persists even when switching to the same VLAN. This may remove the VLAN from the error description.

I currently suspect a misconfiguration on the part of Homeassistant.

danielcherubini, 2025-03-03:
@thiesch yeah, i'm currently leaning towards something on your end too.. since the error log should def say SOMETHING, which it isnt.

https://github.com/danielcherubini/elegoo-homeassistant/blob/main/custom_components/elegoo_printer/config_flow.py

That's the part of the code that you're hitting. As you can see, there's a lot of logging around it. I cover as many errors as i can, including a generic one. On top of that, as you can see on the next link, the call that the config_flow does to test the ip address has a log statement at the start before it does anything.

https://github.com/danielcherubini/elegoo-homeassistant/blob/64e78d74028131733066678b09943d42b138c054/custom_components/elegoo_printer/elegoo_sdcp/elegoo_printer.py#L140

That's kind of telling me that for whatever reason, something else is at play here.

closed 2025-03-21 by danielcherubini