danielcherubini/elegoo-homeassistant #33: support for satrun 3 12k?

ecto1a2003, 2025-03-19:
### Checklist

- [x] I have filled out the template to the best of my ability.
- [x] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [x] This issue is not a duplicate feature request of [previous feature requests](https://github.com/ludeeus/elegoo_printer/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

i think this tool could work with a saturn 3 12k as is seems to have sdcp under the hood somewhere

[image]
[image]

### Describe the solution you'd like

saturn 3 12k support

### Describe alternatives you've considered

none

### Additional context

https://github.com/vvuk/cassini
https://www.reddit.com/r/ElegooSaturn/comments/1edz9nt/saturn_3_ultra_to_home_assistant/

seem to point to some solutions

ecto1a2003, 2025-03-19:
i am available to test

ecto1a2003, 2025-03-19:
status
/sdcp/status/e72274bbee840100
````
{"Id":"f25273b12b094c5a8b9513a30ca60049","Data":{"Status":{"CurrentStatus":0,"PreviousStatus":0,"PrintInfo":{"Status":0,"CurrentLayer":0,"TotalLayer":0,"CurrentTicks":0,"TotalTicks":0,"ErrorNumber":0,"Filename":""},"FileTransferInfo":{"Status":0,"DownloadOffset":0,"CheckOffset":0,"FileTotalSize":0,"Filename":""}},"MainboardID":"e72274bbee840100","TimeStamp":66133}}
````
/sdcp/attributes/e72274bbee840100
````
{"Id":"f25273b12b094c5a8b9513a30ca60049","Data":{"Attributes":{"Name":"Saturn3Ultra","MachineName":"ELEGOO Saturn 3 Ultra","ProtocolVersion":"V1.0.0","FirmwareVersion":"V1.5.4","Resolution":"11520x5120","MainboardIP":"192.168.50.33","MainboardID":"e72274bbee840100","SDCPStatus":1,"LocalSDCPAddress":"tcp://192.168.50.19:1883","SDCPAddress":"","Capabilities":["FILE_TRANSFER","PRINT_CONTROL"]},"MainboardID":"e72274bbee840100","TimeStamp":66032}}
````

danielcherubini, 2025-03-19:
> status /sdcp/status/e72274bbee840100
>
> ``` [...]

This looks like the same data structure. That's great. 

What happens when you add the printer using this integration?

ecto1a2003, 2025-03-19:
Errors out and ha shiws a aiohttp error

ecto1a2003, 2025-03-19:
Aio is the home assistant stack

[image]

ecto1a2003, 2025-03-19:
Aio is the home assistant stack, ill get one in a second

danielcherubini, 2025-03-19:
im trying to think on how to solve this.

Firstly i think during onboarding there should be some kind of question, asking if its an older device or newer device. That will enable the integration to switch between the methods for communicating.

Secondly, I think what makes the most sense, is that this is a mqtt client, which means you'd have to set it up to connect to your mqtt broker, then subscribe to the topics.

ecto1a2003, 2025-03-19:
I currently have it populated in mqtt and am reading that data, cant push anything or send commands

danielcherubini, 2025-03-19:
Right then so it might actually be quite easy then. Just have the integration point to the broker, subscribe to the topics and then I can hook it into all the existing models I've made.

ecto1a2003, 2025-03-19:
Super doable then, matbe Better to build the entities with the integration cuz i had to manually build what i have in the configuration yaml

ecto1a2003, 2025-03-20:
heres the error log

````
Logger: aiohttp.server
Source: /usr/local/lib/python3.13/site-packages/aiohttp/web_protocol.py:451
First occurred: 7:57:38 PM (1 occurrences)
Last logged: 7:57:38 PM

Error handling request from 192.168.50.56
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/site-packages/aiohttp/web_protocol.py", line 480, in _handle_request
    resp = await request_handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/aiohttp/web_app.py", line 569, in _handle
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/aiohttp/web_middlewares.py", line 117, in impl
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/security_filter.py", line 92, in security_filter_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/forwarded.py", line 83, in forwarded_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/request_context.py", line 26, in request_context_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/ban.py", line 86, in ban_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/auth.py", line 242, in auth_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/headers.py", line 32, in headers_middleware
    response = await handler(request)
               ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/http.py", line 73, in handle
    result = await handler(request, **request.match_info)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/decorators.py", line 81, in with_admin
    return await func(self, request, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/config/config_entries.py", line 232, in post
    return await super().post(request, flow_id)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/data_validator.py", line 74, in wrapper
    return await method(view, request, data, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/data_entry_flow.py", line 121, in post
    result = await self._flow_mgr.async_configure(flow_id, data)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 340, in async_configure
    result = await self._async_configure(flow_id, user_input)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 387, in _async_configure
    result = await self._async_handle_step(
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        flow, cur_step["step_id"], user_input
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 543, in _async_handle_step
    result = await self.async_finish_flow(flow, result.copy())
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 1649, in async_finish_flow
    title=result["title"],
          ~~~~~~^^^^^^^^^
KeyError: 'title'
````