danielcherubini/elegoo-homeassistant #35: Centauri Carbon

paranoidi, 2025-04-22:
### System Health details

N/A

### Checklist

- [x] I have enabled debug logging for my installation.
- [x] I have filled out the issue template to the best of my ability.
- [x] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [x] This issue is not a duplicate issue of any [previous issues](https://github.com/ludeeus/elegoo_printer/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

Hi, awesome work!

I'm testing on centauri carbon and things mostly seem to work but there are some issues I've identified.

- sensor.fdm_percent_complete is stuck at 0
- sensor.fdm_remaining_print_time is 0.001894708496387223, it should be about 1h 53 minutes
- sensor.fdm_current_print_time value is similarly non-sensical 0.0006561927487733353
- sensor.fdm_total_print_time is 0.002554, it should be 2h 33min
- sensor.fdm_box_target_temp is unavailable, but I don't think there is any box temp target anyway. Only for nozzle and bed.
- values of sensor.fdm_print_status and sensor.fdm_print_error are in uppercase which is not following HA conventions.

I would also encourage changing the unit to minutes, hours are quite inconvenient for any automation.

### Reproduction steps

N/A

### Debug logs

```text
N/A
```

paranoidi, 2025-04-22:
Since the project & codebase was so clean I was able to easily capture debug information from the `debug.py`. Note that the CurrentLayer is stuck at 0 in the Elegoo WUI as well for this print.

danielcherubini, 2025-04-22:
since the majoity of the work would be in the models, and also the `definitions.py` file... If you want you could make those changes and open a PR... that way we can do this faster since I don't have your printer and would be easier if you debugged it locally.

danielcherubini, 2025-04-22:
Thirdly, I'll carve out some time tonight, to give you time to play with your printer :D

danielcherubini, 2025-04-22:
@paranoidi https://github.com/danielcherubini/elegoo-homeassistant/pull/36

Give this a review if you can and want...

paranoidi, 2025-05-01:
Looking good, merge them in and do a release and I'll test them out. Or if there's some easy way to test them otherwise I'm open for that too.

danielcherubini, 2025-05-02:
@paranoidi its released now in 0.7.0

danielcherubini, 2025-05-02:
if its still happening, reopen this issue.

danielcherubini, 2025-05-02:
@paranoidi I'm re-opening this issue.

in debug.py make your main func look like this

```python
async def main() -> None:
    """Declare Main function for debugging purposes."""
    stop_event = asyncio.Event()
    try:
        printer_ip = os.getenv("PRINTER_IP", "10.0.0.212")
        elegoo_printer = ElegooPrinterClient(printer_ip, logger)
        printer = elegoo_printer.discover_printer()
        if printer:
            connected = await elegoo_printer.connect_printer()
            if connected:
                logger.debug("Polling Started")
                await asyncio.sleep(2)
                # elegoo_printer.set_printer_video_stream(toggle=False)
                # elegoo_printer.get_printer_attributes()
                while not stop_event.is_set():  # noqa: ASYNC110
                    elegoo_printer.get_printer_status()
                    await asyncio.sleep(2)
        else:
            logger.exception("No printers discovered.")
    except asyncio.CancelledError:
        stop_event.set()

```

note how `get_printer_status()` has been moved into the while loop.. do that, and then run `make debug`

danielcherubini, 2025-05-02:
I am curious what your status output is...

danielcherubini, 2025-05-02:
(It's a bit harder to do this on a resin printer.. because im using resin and im wearing out the LCD everytime i run a print :D)

danielcherubini, 2025-05-02:
I decided to start a print, and its looking fine to me

[image]

```json
{
     "Status": {
          "CurrentStatus": [
               1
          ],
          "PrintScreen": 202464.8604933519,
          "ReleaseFilm": 5137,
          "TempOfUVLED": 42.6329345703125,
          "TimeLapseStatus": 0,
          "PrintInfo": {
               "Status": 4,
               "CurrentLayer": 2,
               "TotalLayer": 908,
               "CurrentTicks": 101904,
               "TotalTicks": 7465833,
               "ErrorNumber": 0,
               "Filename": "1-24 - Dredd.goo",
               "TaskId": "9a53ea84-279d-11f0-9ee7-40f4c926a324"
          }
     },
     "MainboardID": "05dc462d920e0100",
     "TimeStamp": 1746222062,
     "Topic": "sdcp/status/05dc462d920e0100"
}
```

paranoidi, 2025-05-03:
[image]

[image]

Switched unit to ms as requested.

Debug logs with modified debug.py

```

2025-05-03 20:33:25.183 | INFO     | custom_components.elegoo_printer.elegoo_sdcp.elegoo_printer:discover_printer:140 - Starting printer discovery at 192.168.x.x
2025-05-03 20:33:25.270 | INFO     | custom_components.elegoo_printer.elegoo_sdcp.elegoo_printer:_save_discovered_printer:177 - Discovered: FDM (192.168.x.x)
2025-05-03 20:33:25.271 | DEBUG    | custom_components.elegoo_printer.elegoo_sdcp.elegoo_printer:discover_printer:158 - Discovery done.
2025-05-03 20:33:25.271 | INFO     | custom_components.elegoo_printer.elegoo_sdcp.elegoo_printer:connect_printer:194 - Connecting to: FDM
2025-05-03 20:33:25.271 | INFO     | custom_components.elegoo_printer.elegoo_sdcp.elegoo_printer:ws_connected_handler:202 - Connected to: FDM
2025-05-03 20:33:27.658 | INFO     | custom_components.elegoo_printer.elegoo_sdcp.elegoo_printer:connect_printer:235 - Connected to FDM
2025-05-03 20:33:27.658 | DEBUG    | __main__:main:29 - Polling Started
2025-05-03 20:33:27.795 | DEBUG    | custom_components.elegoo_printer.elegoo_sdcp.elegoo_printer:_status_handler:283 - status >> 
{
     "Status": {
          "CurrentStatus": [
               1
          ],
          "TimeLapseStatus": 0,
          "PlatFormType": 0,
          "TempOfHotbed": 55.5071776655175,
          "TempOfNozzle": 220.0254057077532,
          "TempOfBox": 28.8793142139421,
          "TempTargetHotbed": 55,
          "TempTargetNozzle": 220,
          "TempTargetBox": 0,
          "CurrenCoord": "101.81,191.49,0.37",
          "CurrentFanSpeed": {
               "ModelFan": 0,
               "AuxiliaryFan": 0,
               "BoxFan": 100
          },
          "ZOffset": 1e-14,
          "LightStatus": {
               "SecondLight": 1,
               "RgbLight": [
                    0,
                    0,
                    0
               ]
          },
          "PrintInfo": {
               "Status": 13,
               "CurrentLayer": 0,
               "TotalLayer": 183,
               "CurrentTicks": 143.25871440093033,
               "TotalTicks": 5071,
               "Filename": "ECC_0.4_SanderBox-Base_PLA0.2_1h25m.gcode",
               "TaskId": "adc6c2bb-301c-4ca7-9a61-682747d0ac57",
               "PrintSpeedPct": 100,
               "Progress": 2
          }
     },
     "MainboardID": "045c14910147017000002c0000000000",
     "TimeStamp": 1746293607,
     "Topic": "sdcp/status/045c14910147017000002c0000000000"
}
2025-05-03 20:33:29.660 | DEBUG    | custom_components.elegoo_printer.elegoo_sdcp.elegoo_printer:_send_printer_cmd:115 - printer << 
{
    "Id": "979d4C788A4a78bC777A870F1A02867A",
    "Data": {
        "Cmd": 0,
        "Data": {},
        "RequestID": "64a30eb02ba6b9e2",
        "MainboardID": "045c14910147017000002c0000000000",
[... 274 lines cut ...]
          "LightStatus": {
               "SecondLight": 1,
               "RgbLight": [
                    0,
                    0,
                    0
               ]
          },
          "PrintInfo": {
               "Status": 13,
               "CurrentLayer": 0,
               "TotalLayer": 183,
               "CurrentTicks": 149.65743894595653,
               "TotalTicks": 5071,
               "Filename": "ECC_0.4_SanderBox-Base_PLA0.2_1h25m.gcode",
               "TaskId": "adc6c2bb-301c-4ca7-9a61-682747d0ac57",
               "PrintSpeedPct": 100,
               "Progress": 2
          }
     },
     "MainboardID": "045c14910147017000002c0000000000",
     "TimeStamp": 1746293613,
     "Topic": "sdcp/status/045c14910147017000002c0000000000"
}
```

paranoidi, 2025-05-03:
In case you're interested. This is the output when printer is preparing the printing (ie. heating up and possibly already doing bed leveling).

It'd be cool to have 'preparing' as status.

```
2025-05-03 20:26:08.544 | DEBUG    | custom_components.elegoo_printer.elegoo_sdcp.elegoo_printer:_status_handler:283 - status >> 
{
     "Status": {
          "CurrentStatus": [
               1
          ],
          "TimeLapseStatus": 0,
          "PlatFormType": 0,
          "TempOfHotbed": 60.59424733423772,
          "TempOfNozzle": 140.0407053288907,
          "TempOfBox": 27.91984861896628,
          "TempTargetHotbed": 60,
          "TempTargetNozzle": 140,
          "TempTargetBox": 0,
          "CurrenCoord": "246.00,65.20,2.00",
          "CurrentFanSpeed": {
               "ModelFan": 0,
               "AuxiliaryFan": 0,
               "BoxFan": 0
          },
          "ZOffset": 1e-14,
          "LightStatus": {
               "SecondLight": 1,
               "RgbLight": [
                    0,
                    0,
                    0
               ]
          },
          "PrintInfo": {
               "Status": 20,
               "CurrentLayer": 0,
               "TotalLayer": 0,
               "CurrentTicks": 0,
               "TotalTicks": 0,
               "Filename": "",
               "TaskId": "",
               "PrintSpeedPct": 100,
               "Progress": 0
          }
     },
     "MainboardID": "045c14910147017000002c0000000000",
     "TimeStamp": 1746293168,
     "Topic": "sdcp/status/045c14910147017000002c0000000000"
}
2025-05-03 20:26:10.382 | DEBUG    | custom_components.elegoo_printer.elegoo_sdcp.elegoo_printer:_send_printer_cmd:115 - printer << 
{
    "Id": "979d4C788A4a78bC777A870F1A02867A",
    "Data": {
        "Cmd": 0,
        "Data": {},
        "RequestID": "67598f5d50f361c2",
        "MainboardID": "045c14910147017000002c0000000000",
        "TimeStamp": 1746293170,
        "From": 0
    },
    "Topic": "sdcp/request/045c14910147017000002c0000000000"
}
2025-05-03 20:26:10.540 | DEBUG    | custom_components.elegoo_printer.elegoo_sdcp.elegoo_printer:_response_handler:274 - response >> 
{
     "Id": "979d4C788A4a78bC777A870F1A02867A",
     "Data": {
          "Cmd": 0,
          "Data": {
               "Ack": 0
[... 597 lines cut ...]
               "RgbLight": [
                    0,
                    0,
                    0
               ]
          },
          "PrintInfo": {
               "Status": 20,
               "CurrentLayer": 0,
               "TotalLayer": 0,
               "CurrentTicks": 0,
               "TotalTicks": 0,
               "Filename": "",
               "TaskId": "",
               "PrintSpeedPct": 100,
               "Progress": 0
          }
     },
     "MainboardID": "045c14910147017000002c0000000000",
     "TimeStamp": 1746293182,
     "Topic": "sdcp/status/045c14910147017000002c0000000000"
}
```

I think `PrintInfo.Status` is 20 for preparing. Just a wild guess looking at these two outputs.

paranoidi, 2025-05-03:
Additionally TempOfHotbed could be exposed as entity to HA.

danielcherubini, 2025-05-03:
I think yes, all the extra temps and stuff we can for sure expose.. it's kind of annoying that they have not updated the spec with those things.. but whatever.. reverse engineering is fun.

Now looking at this json

```json
"Status": {
      "CurrentStatus": [
           1
      ],
      "TimeLapseStatus": 0,
      "PlatFormType": 0,
      "TempOfHotbed": 55.5071776655175,
      "TempOfNozzle": 220.0254057077532,
      "TempOfBox": 28.8793142139421,
      "TempTargetHotbed": 55,
      "TempTargetNozzle": 220,
      "TempTargetBox": 0,
      "CurrenCoord": "101.81,191.49,0.37",
      "CurrentFanSpeed": {
           "ModelFan": 0,
           "AuxiliaryFan": 0,
           "BoxFan": 100
      },
      "ZOffset": 1e-14,
      "LightStatus": {
           "SecondLight": 1,
           "RgbLight": [
                0,
                0,
                0
           ]
      },
      "PrintInfo": {
           "Status": 13,
           "CurrentLayer": 0,
           "TotalLayer": 183,
           "CurrentTicks": 143.25871440093033,
           "TotalTicks": 5071,
           "Filename": "ECC_0.4_SanderBox-Base_PLA0.2_1h25m.gcode",
           "TaskId": "adc6c2bb-301c-4ca7-9a61-682747d0ac57",
           "PrintSpeedPct": 100,
           "Progress": 2
      }
 },
```

Looking at PrintInfo.

```json
 "CurrentTicks": 143.25871440093033,
 "TotalTicks": 5071,
```

5071 - 143 = 4928

I think that's SECONDS not minutes....

danielcherubini, 2025-05-03:
```json
{
     "Attributes": {
          "Name": "FDM Printer",
          "MachineName": "REFERENCE",
          "BrandName": "CHITU",
          "ProtocolVersion": "V3.0.0",
          "FirmwareVersion": "V0.0.0",
          "XYZsize": "218.88x128.88x220",
          "MainboardIP": "192.168.x.x",
          "MainboardMAC": "e8:5c:5f:67:58:d9",
          "MainboardID": "045c14910147017000002c0000000000",
          "SDCPStatus": 0,
          "MaximumCloudSDCPSercicesAllowed": 0,
          "NumberOfCloudSDCPServicesConnected": 0,
          "NumberOfVideoStreamConnected": 0,
          "MaximumVideoStreamAllowed": 4,
          "NetworkStatus": "wlan",
          "UsbDiskStatus": 1,
          "Capabilities": [
               "FILE_TRANSFER",
               "PRINT_CONTROL",
               "VIDEO_STREAM"
          ],
          "SupportFileType": [
               "gcode"
          ],
          "DevicesStatus": {
               "SgStatus": 1,
               "ZMotorStatus": 1,
               "XMotorStatus": 1,
               "YMotorStatus": 1
          },
          "CameraStatus": 1,
          "RemainingMemory": 6437294080,
          "TLPNoCapPos": 0,
          "TLPStartCapPos": 0,
          "TLPInterLayers": 0
     },
     "MainboardID": "045c14910147017000002c0000000000",
     "TimeStamp": 1745315902,
     "Topic": "sdcp/attributes/045c14910147017000002c0000000000"
}
```

Here's your Attributes object. That's annoying here.. is that the Machine Name is `"REFERENCE"` ... also the Firmware version is `V0.0.0`

But .. if we wanted to do something were we said `if MachineName === "CENTAURI CARBON` { use seconds } ELSE { use miliseconds }` or something.. then the machine name, or litteraly any name SHOULD be consistent, and say the right name.. i dunno

paranoidi, 2025-05-03:
If the api is not reliable (at the moment at least) user could select the printer model instead. Or more automated way would be to infer the model from presence or absence of certain attributes.

danielcherubini, 2025-05-03:
Of course!!! have a question at the startup.. I think thats the perfect solution

dvbit, 2025-05-08:
Uhm. I am stuck at ip address in config dialog. Any advice?

fhu667, 2025-05-08:
Downgrade to the previous version, enter your IP address, and then you can upgrade afterwards.

dvbit, 2025-05-08:
just tried (v6). no success....
removed and reinstalled older version

fhu667, 2025-05-08:
0.6.0 was what I had to use to get my IP entered.  Then I was able to go to 0.7.0.

danielcherubini, 2025-05-09:
Is there a bug in 0.7.0 when entering ip address? Is that what im reading?

danielcherubini, 2025-05-09:
@dvbit what error did you see?

dvbit, 2025-05-09:
> @dvbit what error did you see? 

I was seeing an 'unknown error'
After rebooting today config went through

Big thanks! Grazie 
My CC is now visible in ha :-)if i can help with testing lmk

danielcherubini, 2025-05-09:
fixed in v0.7.1.

Also i will try and find some time to add the config option into the printer... to give a boolean to set "this printer uses seconds instead of miliseconds" or something better

danielcherubini, 2025-05-19:
@jtenniswood we know. Currently working on that. Seems that specific printer deviates from their own spec. It sends seconds instead of milliseconds.

danielcherubini, 2025-05-22:
What we're looking for is that the ticks are larger numbers, now milliseconds, so if you divide that by 1000 you get seconds, etc etc.

Supreme-Bob, 2025-05-31:
@cannondale0815 I got that when I had only allowed TCP/80(web interface) TCP/3031(Cam Stream but this integration doesn't use it), did some packet sniffing and found when you add the printer it also uses UDP/3000(Guessing this is some discovery port) and TCP/3030(Guessing this is the API port)

danielcherubini, 2025-05-31:
> I installed version v0.7.1 of this integration, but when I try to add it with my Centauri Carbon's local IP address, I am getting "Unknown error occurred.". Is that version supposed to already work with the Centauri Carbon?
> 
> Logs show "no printer". However, when I go to the same IP address from a web browser, I am successfully seeing the Elegoo web UI.
> 
> ```
> Logger: custom_components.elegoo_printer
> Source: custom_components/elegoo_printer/config_flow.py:41
> integration: Elegoo Printer ([documentation](https://github.com/danielcherubini/elegoo-homeassistant), [issues](https://github.com/danielcherubini/elegoo-homeassistant/issues))
> First occurred: May 26, 2025 at 8:52:18 PM (5 occurrences)
> Last logged: 3:40:44 PM
> 
> Exception: No Printer
> 
> The above exception was the direct cause of the following exception:
> 
> Traceback (most recent call last):
>   File "/config/custom_components/elegoo_printer/config_flow.py", line 41, in async_step_user
>     printer = await self._test_credentials(
>               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
>         ip_address=user_input[CONF_IP_ADDRESS],
>         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
>     )
>     ^
>   File "/config/custom_components/elegoo_printer/config_flow.py", line 85, in _test_credentials
>     raise ElegooPrinterClientGeneralError from Exception("No Printer")
> custom_components.elegoo_printer.config_flow.ElegooPrinterClientGeneralError
> ```

Thats very strange, as that's been patched.

danielcherubini, 2025-06-03:
OK i've tested the time fix.. its working.. i just need to wrap up the configuration

Polinias, 2025-06-05:
> Video, yes this will support that. The issue is that the rtsp stream isn't standard, and needs processing which can be CPU intensive. There's a PR open.

Great addon, thank you!

Regarding camera feed.
It is working out of box using MJPEG IP camera integration, so no need to make any magic stuff. 
Stream is available at http://IP-ADDRESS:3031/video. 

Looking forward to fixed times and ability to control printer, through.

danielcherubini, 2025-06-06:
> Regarding camera feed. It is working out of box using MJPEG IP camera integration, so no need to make any magic stuff. Stream is available at http://IP-ADDRESS:3031/video.
> 
> Looking forward to fixed times and ability to control printer, through.

You're talking about the centauri here right? This is the first i've heard that that camera uses an MJPEG stream... the OTHER printers use a non standard RTSP stream.. hence the need for pre-processing it with FFMPEG.

Of course they changed more things without updating the spec .. LMAO

As for the camera testing, that's something i wont be able to test... i can blindly make it, and hope for the best.. but others with the printer will have to test it.

Polinias, 2025-06-06:
> You're talking about the centauri here right?

Yes. 
I just used standard  [MJPEG IP Camera](https://www.home-assistant.io/integrations/mjpeg) integration. 
Just pasted feed address (can be read in CC webinterface by right-clicking on video feed).
No login/pass, no SSL certificate required, just like that.

[image]

[image]

danielcherubini, 2025-06-06:
great.. for more camera stuff please post/follow #12

danielcherubini, 2025-06-06:
I'd love to find out if the latest version has fixed the time issues

Polinias, 2025-06-06:
> I'd love to find out if the latest version has fixed the time issues

It seems so. Formatting could be more readable, but it is working fine. Thank you!

[image]

danielcherubini, 2025-06-06:
Right. Yes. Thats a bug.

danielcherubini, 2025-06-06:
@Polinias did you remember to remove the integration and re-add it, but turn on the centuari carbon option?

danielcherubini, 2025-06-06:
oh you didn't need to delete the integration.... just remove the device 😄 

[image]

Can you try it again, this time disable the "centuari carbon" option ... the reason is.. it seems you minutes are 1000 more than they should be.

danielcherubini, 2025-06-06:
what it is... is the resin printers (and the spec that elegoo made) are in milliseconds. 

but... on the centurai carbon, its "seconds" not milliseconds. 

So what this whole new version does... is converts seconds to millis.. by multiplying by 1000 ... 

https://github.com/danielcherubini/elegoo-homeassistant/blob/main/custom_components/elegoo_printer/elegoo_sdcp/models/status.py#L74

danielcherubini, 2025-06-06:
I'm at a loss, since ... look take that "remaining print time", and divide it by 1000, thats 2:28hours!

Polinias, 2025-06-06:
> v0.8.3 is published.. grab the latest copy and try now with the option on.

Still something is off.
Elapsed print time: 31m 54s  -> 319m (31,9 min with one magitude too much?)
Remaining: 2:05:31 = 125m:31s -> 1255,17s,  ~125,5 minutes * 10.

Differences on seconds can be ignored, for sure it is coming from refreshing differences.

[image]

danielcherubini, 2025-06-06:
OK new version.. grab it..

Polinias, 2025-06-06:
> OK new version.. grab it..

Like a charm. :)
Thank you! :)

[image]

danielcherubini, 2025-06-06:
WHAT THE HELL!?

```python
        if centauri_carbon:
            self.current_ticks = self.current_ticks * 1
            self.total_ticks = self.total_ticks * 1
            self.remaining_ticks = self.remaining_ticks * 1
```

IT WAS x1 ????

jtenniswood, 2025-06-07:
The add-on works really nicely now. I was having some weird issues with old versions that seem to be DDOS the webui, but  once I figured out this was the issue, the update works as expected. Thanks again for the hard work, nice to be able to get notifications when my prints end now!

danielcherubini, 2025-06-07:
I can eventually move my notification automation to a public blueprint... looks similar to this

[image]

danielcherubini, 2025-06-15:
Unit of measurement. Is what you set. 
Native unit of measurement is what you can't see. It's what the setting sets. 

So my suggestion is make sure you're using the latest version, remove the printer, and re-add it.

paranoidi, 2025-06-15:
@danielcherubini  Thanks! That did it. I did try reinstalling whole addon but that did not help. I though I was on latest entities because of added _2 suffix (that I fixed by hand). Some weird leftovers apparently ...

danielcherubini, 2025-06-15:
Yeah there's not much we can do when you go manual. 

Also the add-on is in HACS now, so if you have the custom repository, you can remove it

commit f3cd0bfc (2025-06-06): Adds new config for using seconds instead of miliseconds, fixes #35 (#44)

* Possible fix for issues in #35

* Adds Options Flow

* Update custom_components/elegoo_printer/translations/en.json

* Update custom_components/elegoo_printer/config_flow.py

* bumping ruff version

* fixing issues with linting

* 📝 Add docstrings to `feature/seconds_selection` (#45)

Docstrings generation was requested by @danielcherubini.

* https://github.com/danielcherubini/elegoo-homeassistant/pull/44#issuecomment-2868949386

The following files were modified:

* `custom_components/elegoo_printer/config_flow.py`
 [... cut]