danielcherubini/elegoo-homeassistant #41: Centauri Carbon Unknown Error when Adding Integration

jdmccreight, 2025-05-09:
### System Health details

## System Information

version | core-2025.4.4
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.13.2
os_name | Linux
os_version | 6.12.23-haos
arch | x86_64
timezone | America/Chicago
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
HACS Data | ok
GitHub API Calls Remaining | 5000
Installed Version | 2.0.5
Stage | running
Available Repositories | 1779
Downloaded Repositories | 28

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 15.2
-- | --
update_channel | stable
supervisor_version | supervisor-2025.04.1
agent_version | 1.7.2
docker_version | 28.0.4
disk_total | 457.7 GB
disk_used | 36.5 GB
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
virtualization | 
board | generic-x86-64
supervisor_api | ok
version_api | ok
installed_addons | Nginx Proxy Manager (1.0.1), Studio Code Server (5.19.3), Uptime Kuma (0.13.0), Z-Wave JS UI (4.2.1), Zigbee2MQTT (2.3.0-1), rtl_433 (0.6.0), rtl_433 MQTT Auto Discovery (0.9.0), Mosquitto broker (6.5.0), AirCast (4.2.4), Piper (1.5.2), openWakeWord (1.10.0), Whisper (2.5.0), Music Assistant Server (2.5.2), Hyperion.NG (2.0.16), Advanced SSH & Web Terminal (20.0.2), Spotify Connect (0.14.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 7
-- | --
resources | 21
views | 20
mode | storage

</details>

<details><summary>Network Configuration</summary>

adapters | lo (disabled), eno1 (enabled, default, auto), docker0 (disabled), hassio (disabled), vethb6c5b59 (disabled), veth444b68e (disabled), veth928f9a8 (disabled), vethfbe437b (disabled), veth9416bac (disabled), veth1ad7c28 (disabled), veth12ce4cd (disabled), veth5bea315 (disabled), veth9392cc4 (disabled), vetheb20ec5 (disabled), veth67b2130 (disabled), vethc5462eb (disabled), veth5e08b0d (disabled), veth0b9d024 (disabled), vethabf53b8 (disabled), veth0efe944 (disabled), vethca031fc (disabled)
-- | --
ipv4_addresses | lo (127.0.0.1/8), eno1 (10.10.15.16/24), docker0 (172.30.232.1/23), hassio (172.30.32.1/23), vethb6c5b59 (), veth444b68e (), veth928f9a8 (), vethfbe437b (), veth9416bac (), veth1ad7c28 (), veth12ce4cd (), veth5bea315 (), veth9392cc4 (), vetheb20ec5 (), veth67b2130 (), vethc5462eb (), veth5e08b0d (), veth0b9d024 (), vethabf53b8 (), veth0efe944 (), vethca031fc ()
ipv6_addresses | lo (::1/128), eno1 (fe80::c823:c27a:7376:c12b/64), docker0 (fe80::b872:c0ff:fee0:540c/64), hassio (fe80::1856:eaff:fe16:5c68/64), vethb6c5b59 (fe80::e44f:6eff:fed1:bcd0/64), veth444b68e (fe80::e428:21ff:fe15:f8a6/64), veth928f9a8 (fe80::688e:90ff:feb8:82dd/64), vethfbe437b (fe80::2c6e:8bff:fe3d:b2fa/64), veth9416bac (fe80::3ced:2cff:fec6:d9a/64), veth1ad7c28 (fe80::8028:eaff:fe1c:c337/64), veth12ce4cd (fe80::80a6:a0ff:fe40:2bc2/64), veth5bea315 (fe80::14c1:5fff:fe02:abf6/64), veth9392cc4 (fe80::ec8f:8aff:fe6d:fad8/64), vetheb20ec5 (fe80::f41b:50ff:febb:ce62/64), veth67b2130 (fe80::1c3a:73ff:fe80:5425/64), vethc5462eb (fe80::d8c0:c3ff:fed6:c278/64), veth5e08b0d (fe80::74e2:45ff:fe8d:638b/64), veth0b9d024 (fe80::dc16:b3ff:fea4:d429/64), vethabf53b8 (fe80::26:55ff:fe79:b87b/64), veth0efe944 (fe80::ccd3:a7ff:fe40:6e7d/64), vethca031fc (fe80::ec66:bdff:fe97:7147/64)
announce_addresses | 10.10.15.16, fe80::c823:c27a:7376:c12b

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | April 20, 2025 at 5:51 PM
-- | --
current_recorder_run | May 9, 2025 at 10:47 AM
estimated_db_size | 2340.71 MiB
database_engine | sqlite
database_version | 3.48.0

</details>

<details><summary>Spotify</summary>

api_endpoint_reachable | ok
-- | --

</details>

### Checklist

- [x] I have enabled debug logging for my installation.
- [x] I have filled out the issue template to the best of my ability.
- [x] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [x] This issue is not a duplicate issue of any [previous issues](https://github.com/ludeeus/elegoo_printer/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

Hello, I am trying to add my Elegoo Centauri Carbon to Home Assistant and when I enter the IP Address of the printer I am met with this message "Unknown error occurred." and the log entry below. This Printer is on a different VLAN, but I'm not seeing any blocking log entries in my Firewall. I changed the VLAN to be on the same one as Home Assistant and got the same error message

### Reproduction steps

1. Download Integration
2. Restart Home Assistant
3. Enter the IP Address, end up with an unknown error

### Debug logs

```text
This error originated from a custom integration.

Logger: custom_components.elegoo_printer
Source: custom_components/elegoo_printer/config_flow.py:41
integration: Elegoo Printer (documentation, issues)
First occurred: 10:49:07 AM (1 occurrences)
Last logged: 10:49:07 AM

ElegooPrinterClient.__init__() missing 1 required positional argument: 'logger'
Traceback (most recent call last):
  File "/config/custom_components/elegoo_printer/config_flow.py", line 41, in async_step_user
    printer = await self._test_credentials(
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        ip_address=user_input[CONF_IP_ADDRESS],
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/config/custom_components/elegoo_printer/config_flow.py", line 81, in _test_credentials
    elegoo_printer = ElegooPrinterClient(ip_address)
TypeError: ElegooPrinterClient.__init__() missing 1 required positional argument: 'logger'
```

danielcherubini, 2025-05-09:
Fixed in v0.7.1

closed 2025-05-09 by danielcherubini

jdmccreight, 2025-05-09:
Confirmed working now even with the printer on a different VLAN, thank you!

commit c9f2db0f (2025-05-09): Fixes #41 (#42)

* Fixes #41

* removing envrc

* removing ignore for brands

* renaming

* updating hacs json

* moving debug

* removing content_in_root

* uses older hacs actions