danielcherubini/elegoo-homeassistant #50: Centauri carbon selection, removes uv related sensors

bystand3r, 2025-06-08:
### Checklist

- [x] I have filled out the template to the best of my ability.
- [x] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [x] This issue is not a duplicate feature request of [previous feature requests](https://github.com/ludeeus/elegoo_printer/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

When adding a new printer and we select the Centauri carbon, it creates the device and sensors but then having to disable them as they are not relevant / needed. 

### Describe the solution you'd like

We we select the Centauri carbon in the config flow can that remove any sensors that are related to resin printers. 

### Describe alternatives you've considered

We we select the Centauri carbon in the config flow can that set to disabled any sensors that are related to resin printers. 

### Additional context

Sensors not needed with Centauri 
Release film
Release film max
UV LED temp
UV LED temp max

danielcherubini, 2025-06-08:
Great idea!

danielcherubini, 2025-06-11:
> How can I send a request to the printer to determine what sensors are actually available?

pull this repo, run make

then edit debug.py, put in your ip_address, make sure centuari_carbon is false, then run make debug

Paste in the status object here.. I have a feeling that you might have extra data in that object, that isn't in the spec.

lukemorrigan-pp, 2025-06-13:
```
{
     "Status": {
          "CurrentStatus": [
               0
          ],
          "TimeLapseStatus": 0,
          "PlatFormType": 1,
          "TempOfHotbed": 19.33598534594573,
          "TempOfNozzle": 21.60678156926167,
          "TempOfBox": 19.50148555438209,
          "TempTargetHotbed": 0,
          "TempTargetNozzle": 0,
          "TempTargetBox": 0,
          "CurrenCoord": "0.00,0.00,0.00",
          "CurrentFanSpeed": {
               "ModelFan": 0,
               "AuxiliaryFan": 0,
               "BoxFan": 0
          },
          "ZOffset": 0.0,
          "LightStatus": {
               "SecondLight": 0,
               "RgbLight": [
                    0,
                    0,
                    0
               ]
          },
          "PrintInfo": {
               "Status": 0,
               "CurrentLayer": 0,
               "TotalLayer": 0,
               "CurrentTicks": 0,
               "TotalTicks": 0,
               "Filename": "",
               "TaskId": "",
               "PrintSpeedPct": 100,
               "Progress": 0
          }
     },
     "MainboardID": "406b18910147017000002c0000000000",
     "TimeStamp": 1749827809,
     "Topic": "sdcp/status/406b18910147017000002c0000000000"
}
```

danielcherubini, 2025-06-14:
yes, we can @Polinias mind making a new feature request? Its hard to keep track when they are comments here.

cbarth3, 2025-06-20:
Just installed this integration and found that even though I checked centauri carbon it still showed the Box, UV, and Release Info. Along with that it did not show any Nozzle or Bed temps. Hope this gets added in a future update.

danielcherubini, 2025-06-21:
It's about to get added today :D

danielcherubini, 2025-06-21:
This should be done now in v0.10.0, please remove the printer and re-add it in the integration.

closed 2025-06-21 by danielcherubini