danielcherubini/elegoo-homeassistant #54: Integration causes Elegoo Web UI to stop working

CodeFoodPixels, 2025-06-11:
### System Health details

## System Information

version | core-2025.5.3
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.13.3
os_name | Linux
os_version | 6.12.23-haos
arch | x86_64
timezone | Europe/London
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
HACS Data | ok
GitHub API Calls Remaining | 5000
Installed Version | 2.0.5
Stage | running
Available Repositories | 2016
Downloaded Repositories | 3

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 15.2
-- | --
update_channel | stable
supervisor_version | supervisor-2025.05.3
agent_version | 1.7.2
docker_version | 28.0.4
disk_total | 30.8 GB
disk_used | 7.0 GB
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
virtualization | kvm
board | ova
supervisor_api | ok
version_api | ok
installed_addons | Studio Code Server (5.19.3), File editor (5.8.0), Netlify DNS (0.0.6), NGINX Home Assistant SSL proxy (3.11.1), Zigbee2MQTT Proxy (0.2.0), Advanced SSH & Web Terminal (20.0.2), Home Assistant Google Drive Backup (0.112.1)

</details>

<details><summary>Dashboards</summary>

dashboards | 2
-- | --
resources | 0
views | 0
mode | storage

</details>

<details><summary>Network Configuration</summary>

adapters | lo (disabled), enp0s18 (enabled, default, auto), hassio (disabled), docker0 (disabled), veth562d935 (disabled), vethe6d2222 (disabled), veth8fc4f80 (disabled), veth1d03d68 (disabled), vethd91e29a (disabled), veth1d98f65 (disabled), veth36d2c77 (disabled), veth7b11642 (disabled), veth8cb890f (disabled), veth40bbac6 (disabled), veth2abdbcf (disabled)
-- | --
ipv4_addresses | lo (127.0.0.1/8), enp0s18 (192.168.2.135/24), hassio (172.30.32.1/23), docker0 (172.30.232.1/23), veth562d935 (), vethe6d2222 (), veth8fc4f80 (), veth1d03d68 (), vethd91e29a (), veth1d98f65 (), veth36d2c77 (), veth7b11642 (), veth8cb890f (), veth40bbac6 (), veth2abdbcf ()
ipv6_addresses | lo (::1/128), enp0s18 (fe80::9339:d329:2545:b9af/64), hassio (fe80::b0fb:9dff:fe8d:c634/64), docker0 (fe80::ac94:5bff:fedb:4ac9/64), veth562d935 (fe80::5487:a9ff:fe05:17ca/64), vethe6d2222 (fe80::3413:36ff:fe2d:cddc/64), veth8fc4f80 (fe80::6ce8:eeff:fee4:3b6/64), veth1d03d68 (fe80::547d:87ff:fe2b:7fd6/64), vethd91e29a (fe80::8bf:42ff:fe06:44b3/64), veth1d98f65 (fe80::641b:b3ff:fe1e:2c41/64), veth36d2c77 (fe80::10c7:8cff:fee8:6e3a/64), veth7b11642 (fe80::d8ce:57ff:fe8a:ca18/64), veth8cb890f (fe80::e48d:17ff:fe8f:a83d/64), veth40bbac6 (fe80::34e3:5fff:fe37:5ae6/64), veth2abdbcf (fe80::dc56:5bff:fe59:1275/64)
announce_addresses | 192.168.2.135, fe80::9339:d329:2545:b9af

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 29 April 2025 at 09:32
-- | --
current_recorder_run | 11 June 2025 at 11:37
estimated_db_size | 17.08 MiB
database_engine | sqlite
database_version | 3.48.0

</details>

### Checklist

- [x] I have enabled debug logging for my installation.
- [x] I have filled out the issue template to the best of my ability.
- [x] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [x] This issue is not a duplicate issue of any [previous issues](https://github.com/ludeeus/elegoo_printer/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

If I switch off my Centauri Carbon with this integration enabled, leave it overnight and then turn it back on, I am unable to use the Elegoo web UI on my printer and the integration doesn't provide me any of the information of the printer. The web UI loads, but the websocket returns a 500 and therefore isn't functional.

If I disable this integration and restart Home Assistant, the web UI works again.

My theory is that there's some sort of backlog of connections that hammer it all at once and take up all the available connections on the websocket.

### Reproduction steps

1. Install and configure integration
2. Verify that integration works
3. Verify that web UI still works
4. Turn off printer and leave for a few hours (Haven't tested specifics, but overnight has worked for me)
5. Turn printer back on
6. Access web UI and see failed websocket connections

### Debug logs

```text
.
```

CodeFoodPixels, 2025-06-12:
> I didn't know there was a WebUI for the Centauri .. Also .. I wonder what
> the max number of websockets is.. since this integration should only use 1
> socket.
>
> My guess... is there's only 1 socket per printer. Which is weird, since
> the Resin printers all have more, and i've never hit any limit, with about
> 3-4 clients connected to a single printer.

You can get to the webUI by putting the IP into your browser, not sure if
this is the case for other printers too.

I've had multiple clients connected at once, I've been able to use the
integration and the web UI at the same time and get up-to-date data in both.

CodeFoodPixels, 2025-06-13:
So I've just started opening tabs to see how many websockets the printer can maintain, it seems to handle 4 connections fine. I'll see if I can do any more investigation.

danielcherubini, 2025-06-13:
In the printer spec (which elegoo has stopped updating publically) there is this max number of camera connections... and also a total of number of connections used.

I wonder if theres something like this for websockets.

Can you check the payload 
https://github.com/danielcherubini/elegoo-homeassistant/issues/50#issuecomment-2963091347

Also update that thread with your payload.

cdunn95, 2025-06-13:
> In the printer spec (which elegoo has stopped updating publically) there is this max number of camera connections... and also a total of number of connections used.
>
> I wonder if theres something [...]

Can also add that this is a thing with the CC. 

Elegoo has admitted that they used quite inferior hardware for the board of this printer in an attempt to keep costs down (thus the reason it does not run native Klipper), as such it is extremely easy to overload the board with too many requests. 

One thing that I have considered doing but have yet to get around to is to make a branch of your add-on, but adding in a middle-man Nginx instance that simply points to the printer as a singular, persistent, relay connection to forward to the HA plugin. 

FWIW, I don't believe that it's your plugin in specific that is causing this issue, I think it's a mixture of Elegoo's hardware in the CC, HomeAssistant instantiating multiple instances of the WS connection (not properly disposing of old ones in a timely manner), and this add-on. 

For more info, check out this conversation I was having surrounding this topic in the Elegoo Discord.

danielcherubini, 2025-06-15:
Updating from some discovery @cdunn95 has done.

The `VideoStreams` sensor seems to track both connections of video AND websocket connections. So every WS Connection seems to add to that tally. There's a MaxVideoStreams set to 4. Which when that goes to 4, then the WebUI will not connect anymore. 

I think there's two things at play.
1. That as i've said WS Connections add to the VideoStreams tally
2. The Elegoo WebUI checks that number, and if its greater than the MaxVideoStreams, it wont connect to the websocket.

@cdunn95 is going to check if we can connect the integration more than 4 times, so no video.. just WS Connections .. 

From this I expect
1. Each connection adds 1 to the VideoStreams tally.
2. We can connect more than 4 times

If thats true, then the issue might actually be firmware related. On the Elegoo printer AND the WebUI. There may be a good reason why this is the way it is.. OR this might be a bug.

It's kind of WEIRD that they would have a VideoStreams tally, for both video AND websockets. Why not create a new entity, like the S4U which has 2 MaxVideoStreams, but that's ONLY for video streams. Why not have `VideoStreamsConnected` and `WebSocketConnections`.

judgeu, 2025-06-23:
Thanks for writing this issue ;) 🙇 
Just wanna share my experience. ℹ ♾ 

Besides the WebUI not working anymore (as soon as this integration is active in HA) also OrcaSlicer could not send any jobs.

I tried to disable + remove + restart every single bit.
- removing the entities from the dashboard
- disable the Elegoo entity
- disable the integration
- delete the integration

And restart the printer (in my case Centauri Carbon).
Nothing helped, WebGUI was still not responsive (WebGui was loading but no data came through via websocket).

The only thing that helped was restarting HA. 💡 ✅ 
So I guess this closed all open connections - finally.

WebGUI responsive again 👍 
OrcaSlicer working again 👍 

Hope that helps others until there might be a solution for more parallel connections

bluesk1d, 2025-06-23:
FWIW, I contacted Elegoo support about this issue before realizing it was this integration. This was part of their response:

"Please note that the cc printer is limited to 4 interface connections. Please check whether the interface occupies more than 4 ports; note that the software occupies two ports; if there are too many connected devices, you can ask the customer to turn off the connection with other devices and restart the machine."

The integration _seems_ to eat up all 4 because the web UI and the Device tab of Orca are all unable to get a connection to the printer. I'm normally able to view the printer from multiple places (if the integration isn't running).

danielcherubini, 2025-06-23:
Thanks. Were aware. And it's next up on the docket.

bluesk1d, 2025-06-23:
My man! It looks like you are proxying at the socket layer so it should allow http to upgrade to WS (so it can be used by a standard web browser). Is that the case?

danielcherubini, 2025-06-23:
Yes, thats the case. Right now it might allow for as many different things to connect to it as possible.. we can batten down that hatch later if we want.

BamiPen, 2025-06-24:
What a integration 🙌🏻

Really like your work. 

Does this proxy server functionality also work for home assistant via docker on a synology nas?

Or am I just to early on the party and I hev to wait to a new release 😊?

danielcherubini, 2025-06-26:
Just updating this thread. It's progressing nicely.

Auto-discovery,  websockets, and file uploads are all done. 

File uploads are slow, I had hoped that this part would be faster. But it's 1mb chunks, so there's an overhead of sending it to the printer and back. I don't want to cache everything then send it to the printer. Cause man some files can be really really big, and not everyone has memory to spare. 

I'm going to try and make a queue or something so it can go a little faster, but also keep the memory constraints down. 

I'm also going to add a configuration option to enable or disable the proxy server. As us with resin printers don't need this (for now)

This will also go out in a larger release, probably a 1.0, where it will also have auto discovery.. so instead of manually adding the printer, the home assistant will discover the printer and suggest it if it's not already there.

danielcherubini, 2025-06-26:
If anyone feels comfortable reviewing python. Please do so.

danielcherubini, 2025-06-27:
remove the printer and add it again.

bluesk1d, 2025-06-27:
That did it. It looks like 3030 is the only one listening but I'm not sure how to initiate a new connection via slicer or web browser to allow the upgrade to WS. Do you know what the slicer is expecting? 

192.168.1.6:3030 (HA) Results in a 404 from Orca
192.168.1.254:3030 (actual printer) works in Orca
192.168.1.6 Is refused since it's not listening on 80

tcp        0      0 0.0.0.0:5355            0.0.0.0:*               LISTEN     
tcp        0      0 127.0.0.1:139           0.0.0.0:*               LISTEN     
tcp        0      0 0.0.0.0:111             0.0.0.0:*               LISTEN     
tcp        0      0 127.0.0.1:18554         0.0.0.0:*               LISTEN     
tcp        0      0 0.0.0.0:22              0.0.0.0:*               LISTEN     
tcp        0      0 127.0.0.1:445           0.0.0.0:*               LISTEN     
tcp        0      0 192.168.1.6:40829       0.0.0.0:*               LISTEN     
tcp        0      0 172.30.32.1:5580        0.0.0.0:*               LISTEN     
tcp        0      0 0.0.0.0:8485            0.0.0.0:*               LISTEN     
tcp        0      0 127.0.0.1:58751         0.0.0.0:*               LISTEN     
tcp        0      0 0.0.0.0:4357            0.0.0.0:*               LISTEN     
tcp        0      0 192.168.1.6:139         0.0.0.0:*               LISTEN     
tcp        0      0 0.0.0.0:8884            0.0.0.0:*               LISTEN     
tcp        0      0 0.0.0.0:8883            0.0.0.0:*               LISTEN     
tcp        0      0 172.30.32.1:65127       0.0.0.0:*               LISTEN     
tcp        0      0 127.0.0.1:11984         0.0.0.0:*               LISTEN     
tcp        0      0 127.0.0.1:65127         0.0.0.0:*               LISTEN     
tcp        0      0 192.168.1.6:40000       0.0.0.0:*               LISTEN     
tcp        0      0 192.168.1.6:445         0.0.0.0:*               LISTEN     
tcp        0      0 0.0.0.0:3030            0.0.0.0:*               LISTEN     
tcp        0      0 172.30.32.1:63269       0.0.0.0:*               LISTEN     
tcp        0      0 0.0.0.0:8123            0.0.0.0:*               LISTEN     
tcp        0      0 0.0.0.0:1884            0.0.0.0:*               LISTEN     
tcp        0      0 0.0.0.0:1883            0.0.0.0:*               LISTEN

bluesk1d, 2025-06-27:
The plot thickens. When I tested via nc I saw that 3030 was listening and assumed it was your proxy but it's not. Something in HA is using that port but I'm not sure what it is since it's the main HA python PID. Trying to see if I can find the culprit. Is it possible to add an override for the proxy port for others that can't find/change an existing binding for 3030?

bluesk1d, 2025-06-27:
I think I got tricked by a red herring. The log was getting spammed by your integration saying it couldn't bind 3030 so I assumed it must be something else. When I removed the integration the 3030 binding went away so maybe it's trying to spawn multiple for some reason?

I deleted the printer and completely removed the integration and re-installed it. I no longer get the binding errors but I do get the same behavior as initially reported. This is all I see in the logs about unhandled HTTP requests.

```
2025-06-27 13:59:44.426 WARNING (Thread-4 (_start_servers_in_thread)) [custom_components.elegoo_printer] Received unhandled HTTP request: GET /api/version
2025-06-27 13:59:44.428 WARNING (Thread-4 (_start_servers_in_thread)) [custom_components.elegoo_printer] Received unhandled HTTP request: GET /
2025-06-27 14:00:01.994 DEBUG (MainThread) [custom_components.elegoo_printer] Finished fetching elegoo_printer data in 0.000 seconds (success: True)
2025-06-27 14:00:02.320 WARNING (MainThread) [custom_components.ef_ble.eflib.connection - A0:DD:6C:10:CD:6E] Disconnected from device
2025-06-27 14:00:02.321 WARNING (MainThread) [custom_components.ef_ble.eflib.connection - A0:DD:6C:10:CD:6E] Reconnecting to the device in 10 seconds...
2025-06-27 14:00:03.996 DEBUG (MainThread) [custom_components.elegoo_printer] Finished fetching elegoo_printer data in 0.000 seconds (success: True)
2025-06-27 14:00:04.647 WARNING (Thread-4 (_start_servers_in_thread)) [custom_components.elegoo_printer] Received unhandled HTTP request: GET /api/version
2025-06-27 14:00:04.650 WARNING (Thread-4 (_start_servers_in_thread)) [custom_components.elegoo_printer] Received unhandled HTTP request: GET /
2025-06-27 14:00:05.998 DEBUG (MainThread) [custom_components.elegoo_printer] Finished fetching elegoo_printer data in 0.001 seconds (success: True)
2025-06-27 14:00:07.995 DEBUG (MainThread) [custom_components.elegoo_printer] Finished fetching elegoo_printer data in 0.001 seconds (success: True)
2025-06-27 14:02:41.993 DEBUG (MainThread) [custom_components.elegoo_printer] Finished fetching elegoo_printer data in 0.000 seconds (success: True)
2025-06-27 14:02:43.994 DEBUG (MainThread) [custom_components.elegoo_printer] Finished fetching elegoo_printer data in 0.000 seconds (success: True)
2025-06-27 14:02:45.994 DEBUG (MainThread) [custom_components.elegoo_printer] Finished fetching elegoo_printer data in 0.000 seconds (success: True)
2025-06-27 14:02:47.994 DEBUG (MainThread) [custom_components.elegoo_printer] Finished fetching elegoo_printer data in 0.000 seconds (success: True)
2025-06-27 14:02:49.995 DEBUG (MainThread) [custom_components.elegoo_printer] Finished fetching elegoo_printer data in 0.000 seconds (success: True)
2025-06-27 14:02:51.994 DEBUG (MainThread) [custom_components.elegoo_printer] Finished fetching elegoo_printer data in 0.000 seconds (success: True)
2025-06-27 14:02:52.051 WARNING (Thread-4 (_start_servers_in_thread)) [custom_components.elegoo_printer] Received unhandled HTTP request: GET /api/version
2025-06-27 14:02:52.067 WARNING (Thread-4 (_start_servers_in_thread)) [custom_components.elegoo_printer] Received unhandled HTTP request: GET /
2025-06-27 14:02:53.994 DEBUG (MainThread) [custom_components.elegoo_printer] Finished fetching elegoo_printer data in 0.000 seconds (success: True)
2025-06-27 14:02:54.429 WARNING (Thread-4 (_start_servers_in_thread)) [custom_components.elegoo_printer] Received unhandled HTTP request: GET /api/version
2025-06-27 14:02:54.431 WARNING (Thread-4 (_start_servers_in_thread)) [custom_components.elegoo_printer] Received unhandled HTTP request: GET /
2025-06-27 14:02:55.994 DEBUG (MainThread) [custom_components.elegoo_printer] Finished fetching elegoo_printer data in 0.000 seconds (success: True)

```

danielcherubini, 2025-06-27:
Those debug fetch logs are normal. Because you have debug logging on. 

I have a PR about the binding error message spamming.

bluesk1d, 2025-06-28:
Based on Wireshark and testing with Postman, best I can guess is that Orca expects to go through the HTTP > WS upgrade process and the proxy doesn't allow this.

danielcherubini, 2025-06-28:
That upgrade request should be easy to implement

danielcherubini, 2025-06-28:
https://github.com/danielcherubini/elegoo-homeassistant/releases/tag/v0.13.3

Try this

danielcherubini, 2025-06-28:
I need to make the errors in the logs a bit more quiet. But so far this now also proxies more than before.

suprra, 2025-06-28:
I could connect using home assistant port now, although likely it was just a me problem but I'm glad it's working now, I'm going to private the proxy so you can continue your excellency : ) Thank you

suprra, 2025-06-28:
> [@suprra](https://github.com/suprra) when you say this. You mean in like Orca?

Yep, well it said tested successfully but I’m waiting for my print to finish so I can see if it will do the more complicated requests

bluesk1d, 2025-06-29:
Can confirm the http > ws upgrade flow works properly now and I can use this proxy in HA as well as a standard browser tab. Thanks for your work!

danielcherubini, 2025-06-29:
It's your HA IP. As for a sensor... I guess yeah.

bystand3r, 2025-06-29:
Thanks that’s working flawlessly now :)

danielcherubini, 2025-06-29:
Ok. We can close this thread.