danielcherubini/elegoo-homeassistant #59: A similar project already exists

bushvin, 2025-06-14:
### System Health details

Hey, I happen to have run intro your project, and it is all too similar to mine…

https://github.com/bushvin/hass_chitubox_printer

### Checklist

- [x] I have enabled debug logging for my installation.
- [x] I have filled out the issue template to the best of my ability.
- [x] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [x] This issue is not a duplicate issue of any [previous issues](https://github.com/ludeeus/elegoo_printer/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

N/a

### Reproduction steps

1.n/a
2.
3.
...

### Debug logs

```text
N/a
```

bushvin, 2025-06-14:
Good question.

I am not a (python) developer, neither am I good at coding particularly well. I do know python.
I did write an [sdcp api](https://gitlab.com/bushvin/sdcpapi) based on the SDCP protocol I found [here](https://github.com/cbd-tech/SDCP-Smart-Device-Control-Protocol-V3.0.0/blob/main/SDCP(Smart%20Device%20Control%20Protocol)_V3.0.0_EN.md).

I think it would be logical to join forces. I can't tell how mature your project is, as I cannot get yours running on my setup (my printer sits in a different network than my HA, and I think that is the reason why it is not working).

Judging from your code you know a lot more than me about (home assistant) development. So I could contribute to your code to get similar/same features? And then long term deprecate my project?

danielcherubini, 2025-06-14:
Cool idea. 

Yeah that would be the best. This project is a few months older than yours, is published on HACS, and we're already solving issues with the new FDM printers, which have deviated slightly from the spec. Which as you know.. elegoo loves to do. 

The best ideaight be to use your SDCP. client in this project. It's a much cleaner solution. And would seperate things well. As debugging the client is half the job. Where as this can be like yours and a thin wrapper. 

I liked how you solved the BMP problem with them not sending the right headers. That was the solution I wanted to find time to do. 

As for me. Yeah I'm a professional senior engineer, we also have another who wants to help too. So that's great. 

Oh. If you thought the issues with the camera sucked... Wait till you read the other issue ticket about the Centauri carbon printer and it's max 4 connections total. As you've seen with the S4U camera.. it doesn't free up connections unless you power cycle it... Well.. now same issue, except 4 websocket connections and the slicer is one, the camera is another. 

If you want. Could you solve the image issue?

bushvin, 2025-06-15:
I should be able to manage that 😉
Although, you may want to go through my api library first, I'm not sure everything is done the way it should be...

danielcherubini, 2025-06-15:
I'm going to close this, we can discuss this another way... i'm talking with another via discord, so if you're open to that, we could just make a discord or a slack or something, or maybe theres another option, I'm open to suggestions
I just opened the discussions section of this repo
https://github.com/danielcherubini/elegoo-homeassistant/discussions/60

closed 2025-06-15 by danielcherubini