danielcherubini/elegoo-homeassistant #61: Thumbnail

danielcherubini, 2025-06-15:
### Checklist

- [x] I have filled out the template to the best of my ability.
- [x] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [x] This issue is not a duplicate feature request of [previous feature requests](https://github.com/ludeeus/elegoo_printer/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

Currently there is no thumbnail support.

### Describe the solution you'd like

@bushvin has a great implementation which solves the issue with the header being wrong and the filetype being not great for most browsers.

### Describe alternatives you've considered

n/a

### Additional context

@bushvin if you could solve this that would be great.

bushvin, 2025-06-16:
I cannot read your code.
This is not bad, as it's because of me. I see there is an api and a printer client which are intertwined, probably for good reasons, but not having written the code, I can't wrap my head around it. And then there's `PrinterData` fetched in various places, ... Not having a coder background doesn't make things easier. Give me distributed and containerized environments any day!

Whenever I code, I use this mantra: 

> Always code as if the guy who ends up maintaining your code will be a violent psychopath who knows where you live.
> Code for readability

Do you have a schema of `ElegooDataUpdateCoordinator.data`? or can you tell me which variable I can use to get the path of the thumbnail?

danielcherubini, 2025-06-16:
> Give me distributed and containerized environments any day!
That's what the .devcontainer.json is.

> Do you have a schema of ElegooDataUpdateCoordinator.data? or can you tell me which variable I can use to get the path of the thumbnail?

I'm guessing you don't have any code-typing/typeahead/etc working for you. 

For me, when i would "go to definition" on that class, it goes to `coordinator.py`

[image]

Which you can see here gives you code hints...

[image]

Going to def there takes you to `data.py`

And there you go... `ElegooPrinterData`

you can see that it points to an `api.py`

and ... there you go... it points to the API, and `ElegooPrinterClient` which is the main api class... from there.. you're looking at the `PrinterData` class... and inside that 

[image]

Also check out the previous attempt i abandoned
https://github.com/danielcherubini/elegoo-homeassistant/pull/24/files#diff-dd5ddc08c01564e8cebfb8016369f75fe0f178d2a466071e11799a5f5caa4d8f

https://github.com/danielcherubini/elegoo-homeassistant/pull/24/files

danielcherubini, 2025-06-16:
There's a function already there for you in the `ElegooPrinterClient` 

[image]

danielcherubini, 2025-06-16:
If you want me to do it.. thats cool too... its a good learning experience IMHO

danielcherubini, 2025-06-16:
I've updated the readme with instructions on getting the project running in various ways.