danielcherubini/elegoo-homeassistant #75: Filament Runout

jtenniswood, 2025-06-21:
### Checklist

- [x] I have filled out the template to the best of my ability.
- [x] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [x] This issue is not a duplicate feature request of [previous feature requests](https://github.com/ludeeus/elegoo_printer/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

I would love to be able to see the filament sensor in home assistant, to send a notification when it needs changing.

I had this happen today and couldn't find any error or status data that would be a suitable trigger.

Thanks!

### Describe the solution you'd like

Detailed above

### Describe alternatives you've considered

Detailed above 

### Additional context

Detailed above

danielcherubini, 2025-06-21:
> I had this happen today and couldn't find any error or status data that would be a suitable trigger.

This is the thing im afraid of too... if there's no status or error, its a bit hard for me to see it... there would HAVE to be something.

If you don't mind, clone this repo, edit debug.py adding in your printer IP address, then run the following

```bash
make setup
make debug
```

It will spit out a whole lot of json, I'm curious if there's something in the attributes or status that would indicate a filament runout.. there would have to be.

bystand3r, 2025-06-23:
i feel like a complete simpleton, why is this failing for me. do i need to do something else ? 
Ben Hovell@UNRAIDVM MINGW64 ~/elegoo-homeassistant (main)
$ make setup
curl -LsSf https://astral.sh/uv/install.sh | sh && uv sync --all-extras --dev
downloading uv 0.7.13 x86_64-pc-windows-msvc
no checksums to verify
installing to /c/Users/Ben Hovell/.local/bin
  uv.exe
  uvx.exe
  uvw.exe
everything's installed!
Resolved 117 packages in 3ms
Audited 111 packages in 0.10ms

Ben Hovell@UNRAIDVM MINGW64 ~/elegoo-homeassistant (main)
$ make debug
DEBUG=true python -m debug
Traceback (most recent call last):
  File "C:\Python310\lib\runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Python310\lib\runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "C:\Users\Ben Hovell\elegoo-homeassistant\debug.py", line 9, in <module>
    from custom_components.elegoo_printer.elegoo_sdcp.const import DEBUG
  File "C:\Users\Ben Hovell\elegoo-homeassistant\custom_components\elegoo_printe
r\__init__.py", line 13, in <module>
    from homeassistant.const import Platform
ModuleNotFoundError: No module named 'homeassistant'
make: *** [Makefile:14: debug] Error 1

Ben Hovell@UNRAIDVM MINGW64 ~/elegoo-homeassistant (main)
$

closed 2025-06-23 by danielcherubini