danielcherubini/elegoo-homeassistant #86: Centauri Carbon on different subnet not reachable by this integration

cannondale0815, 2025-06-23:
### System Health details

## System Information

version | core-2025.6.1
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.13.3
os_name | Linux
os_version | 6.6.74-haos-raspi
arch | aarch64
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
HACS Data | ok
GitHub API Calls Remaining | 5000
Installed Version | 2.0.5
Stage | running
Available Repositories | 2039
Downloaded Repositories | 22

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 15.2
-- | --
update_channel | stable
supervisor_version | supervisor-2025.06.1
agent_version | 1.7.2
docker_version | 28.0.4
disk_total | 116.8 GB
disk_used | 68.6 GB
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
virtualization | 
board | rpi4-64
supervisor_api | ok
version_api | ok
installed_addons | AdGuard Home (5.2.10), Assistant Relay (0.7.4), Home Assistant Google Drive Backup (0.112.1), Node-RED (19.0.2), Advanced SSH & Web Terminal (20.0.2), Matter Server (8.0.0), Studio Code Server (5.19.3), Duck DNS (1.19.0), Mosquitto broker (6.5.1), Dnsmasq (1.8.1), Nginx Proxy Manager (1.0.1), MariaDB (2.7.2), Zigbee2MQTT (2.4.0-1), Lovelace Kindle Screensaver (1.0.14), Portainer (2.31.1), AppDaemon (0.17.4), HA Skill (1.0.7), TheengsGateway (1.11.5), Plex Media Server (3.6.4), ESPHome Device Builder (2025.6.0), Orb (1.0.10), Govee to MQTT Bridge (2025.04.13-17d43d72), Music Assistant Server (2.5.5)

</details>

<details><summary>Network Configuration</summary>

adapters | lo (disabled), end0 (enabled, default, auto), wlan0 (disabled), hassio (disabled), docker0 (disabled), vethac618b3 (disabled), vethb8915aa (disabled), veth2ac711d (disabled), vetha7b0356 (disabled), vethf09e84d (disabled), veth20acde6 (disabled), veth26990eb (disabled), vethe805cfd (disabled), veth747f7a4 (disabled), vethc59703b (disabled), veth0219e30 (disabled), veth21a5e37 (disabled), vethba40cca (disabled), veth9dcf550 (disabled)
-- | --
ipv4_addresses | lo (127.0.0.1/8), end0 (192.168.10.10/24), wlan0 (192.168.10.170/24), hassio (172.30.32.1/23), docker0 (172.30.232.1/23), vethac618b3 (), vethb8915aa (), veth2ac711d (), vetha7b0356 (), vethf09e84d (), veth20acde6 (), veth26990eb (), vethe805cfd (), veth747f7a4 (), vethc59703b (), veth0219e30 (), veth21a5e37 (), vethba40cca (), veth9dcf550 ()
ipv6_addresses | lo (::1/128), end0 (fd1c:5228:49df:9576:f62b:1c33:2791:d71b/64, fe80::c479:d8de:1a5d:9f25/64), wlan0 (fe80::b1f5:cae4:193c:cc37/64), hassio (fe80::902e:d1ff:fe97:c40d/64), docker0 (fe80::74f0:b3ff:fe4c:a42e/64), vethac618b3 (fe80::ac95:e9ff:fe5f:d164/64), vethb8915aa (fe80::14a9:eeff:fe7d:8d4/64), veth2ac711d (fe80::5886:7fff:feb7:8c98/64), vetha7b0356 (fe80::3870:25ff:fe48:f00e/64), vethf09e84d (fe80::e448:68ff:fe31:93b3/64), veth20acde6 (fe80::9885:38ff:fe6d:552/64), veth26990eb (fe80::7c85:31ff:fe2b:f232/64), vethe805cfd (fe80::f0dd:cff:feba:1510/64), veth747f7a4 (fe80::b86e:caff:fe8e:202a/64), vethc59703b (fe80::bc2f:35ff:fe18:8607/64), veth0219e30 (fe80::48c1:43ff:fe96:1db5/64), veth21a5e37 (fe80::5006:8cff:fe5b:f3e8/64), vethba40cca (fe80::b4a5:c7ff:fe2c:fb34/64), veth9dcf550 (fe80::a835:d8ff:fe17:b2a4/64)
announce_addresses | 192.168.10.10, fd1c:5228:49df:9576:f62b:1c33:2791:d71b, fe80::c479:d8de:1a5d:9f25

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | June 12, 2025 at 2:36 PM
-- | --
[... 95 lines cut ...]
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/data_validator.py", line 74, in wrapper
    return await method(view, request, data, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/data_entry_flow.py", line 121, in post
    result = await self._flow_mgr.async_configure(flow_id, data)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 336, in async_configure
    result = await self._async_configure(flow_id, user_input)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 383, in _async_configure
    result = await self._async_handle_step(
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        flow, cur_step["step_id"], user_input
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 487, in _async_handle_step
    result: _FlowResultT = await getattr(flow, method)(user_input)
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/elegoo_printer/config_flow.py", line 113, in async_step_user
    printer_object.ip_address = user_input[CONF_IP_ADDRESS]
    ^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'ip_address' and no __dict__ for setting new attributes
```

cannondale0815, 2025-06-23:
Incredibly fast response -- let me give it a whirl and I shall report back shortly! :)

danielcherubini, 2025-06-23:
try main first... you should see something similar to this

[image]

if you can see "Discovered <printername>" ... thats progress.. if not.. then yeah

cannondale0815, 2025-06-23:
Sorry, took me a while as I had to install the Linux subsystem on my Win 11 laptop first (no physical Linux box available to me at the moment). Following your initial suggestions, I edited the debug.py file to include my printer IP address and the elegoo_printer.py to increase the timeout to ten seconds. In each case, I get the following:

```
~/ha-elegoo/elegoo-homeassistant$ make debug
--> Starting the application in DEBUG mode...
2025-06-23 18:31:41.085 | INFO     | custom_components.elegoo_printer.elegoo_sdcp.elegoo_printer:discover_printer:148 - Starting printer discovery at 192.168.12.229
2025-06-23 18:31:51.093 | WARNING  | custom_components.elegoo_printer.elegoo_sdcp.elegoo_printer:discover_printer:160 - Printer discovery timed out.
2025-06-23 18:31:51.094 | ERROR    | __main__:main:38 - No printers discovered.
```

Haven't tried your new branch yet -- am a bit confused how to pull it -- what's the branch name?

cannondale0815, 2025-06-23:
Hmmm, with the new "better_broadcast" branch, I am getting the following:

```
~/ha-elegoo/elegoo-homeassistant$ make debug
--> Starting the application in DEBUG mode...
2025-06-23 18:36:41.133 | INFO     | custom_components.elegoo_printer.elegoo_sdcp.elegoo_printer:discover_printer:174 - Starting printer discovery across all local subnets.
2025-06-23 18:36:41.134 | INFO     | custom_components.elegoo_printer.elegoo_sdcp.elegoo_printer:discover_printer:178 - Will attempt discovery on: ['172.27.223.255']
2025-06-23 18:36:41.134 | DEBUG    | custom_components.elegoo_printer.elegoo_sdcp.elegoo_printer:discover_printer:196 - Sending discovery packet to 172.27.223.255
2025-06-23 18:36:51.142 | DEBUG    | custom_components.elegoo_printer.elegoo_sdcp.elegoo_printer:discover_printer:215 - No response from broadcast to 172.27.223.255, trying next.
2025-06-23 18:36:51.142 | WARNING  | custom_components.elegoo_printer.elegoo_sdcp.elegoo_printer:discover_printer:224 - Discovery finished. No printers found on any interface.
2025-06-23 18:36:51.143 | ERROR    | __main__:main:39 - No printers discovered.
NoneType: None
```

danielcherubini, 2025-06-23:
If that doesn't work... try this branch `feature/better_discovery`

danielcherubini, 2025-06-23:
the latter is a re-work of the discovery to use the auto-discovery... you wont even need to enter your ip in.. it should find it on the broadcast

cannondale0815, 2025-06-23:
Thank you for all this. This is what I'm getting with the better_discovery branch:

```
~/ha-elegoo/elegoo-homeassistant$ make debug
--> Starting the application in DEBUG mode...
2025-06-23 18:36:41.133 | INFO     | custom_components.elegoo_printer.elegoo_sdcp.elegoo_printer:discover_printer:174 - Starting printer discovery across all local subnets.
2025-06-23 18:36:41.134 | INFO     | custom_components.elegoo_printer.elegoo_sdcp.elegoo_printer:discover_printer:178 - Will attempt discovery on: ['172.27.223.255']
2025-06-23 18:36:41.134 | DEBUG    | custom_components.elegoo_printer.elegoo_sdcp.elegoo_printer:discover_printer:196 - Sending discovery packet to 172.27.223.255
2025-06-23 18:36:51.142 | DEBUG    | custom_components.elegoo_printer.elegoo_sdcp.elegoo_printer:discover_printer:215 - No response from broadcast to 172.27.223.255, trying next.
2025-06-23 18:36:51.142 | WARNING  | custom_components.elegoo_printer.elegoo_sdcp.elegoo_printer:discover_printer:224 - Discovery finished. No printers found on any interface.
2025-06-23 18:36:51.143 | ERROR    | __main__:main:39 - No printers discovered.
NoneType: None
```

Just to add, I can go to http://192.168.12.229 and see the web UI of the printer just fine. From the same laptop that I am running the Ubuntu subsystem on right now. Strange.

danielcherubini, 2025-06-23:
I think its a network problem

but before we get there

make a file called `.wslconfig` in your `%USERPROFILE%` directory. paste this in
```toml
[wsl2]
networkingMode=mirrored
```

then 

```powershell
wsl --shutdown
```

open wsl again, and now you've got mirrored network mode, which fixes any broadcast issue you get inside the wsl container (you will, i know because i dev in wsl2)

Also

[image]

danielcherubini, 2025-06-23:
I know you can see it via HTTP.... but thats HTTP.. your router might be blocking non-http traffic

danielcherubini, 2025-06-23:
i just added a test file to that branch you're on... pull the changes and run `uv run test.py`

This strips it all back.. just the socket connection, if that works .. ok.. if that doesn't then perhaps something is blocking this specific traffic on your network

cannondale0815, 2025-06-23:
Yeah, I'm getting this now:

```
~/ha-elegoo/elegoo-homeassistant$ uv run test.py 192.168.12.229
INFO: Using IP address from command line: 192.168.12.229
----------------------------------------
Attempting to send direct message to printer...
  - Destination: 192.168.12.229:3000
  - Message:     b'M99999'
  - Timeout:     5.0 seconds
----------------------------------------
STEP 1: Sending message...
       ...Message sent successfully.

STEP 2: Waiting for a reply...

❌ --- FAILURE: TIMEOUT --- ❌
The operation timed out because no response was received from the printer.

This could mean:
  1. The IP address is wrong or the printer is offline.
  2. A firewall on your computer (Windows Defender) is blocking the incoming UDP packet.
  3. The printer is programmed to ignore direct messages and only responds to broadcasts.
```

And I will say (and perhaps I should have mentioned it before), the printer is on an isolated IoT network, but all devices on my regular LAN can talk to all IoT devices, just not the other way around. So that's probably the issue.

I'm running an OpenWRT router and had the same issues with HomeKit devices. I had to create custom rules for the discovery. Let me lift all isolation restriction for a moment and give it a try. 

Would you happen to know what exactly needs to be opened up for the discovery process?

cannondale0815, 2025-06-23:
I think I figured it out -- I created the following two rules:

[image]

And now the discovery works!

danielcherubini, 2025-06-23:
To explain further... you have to do the discovery call, it replies with two things that are really important, the ip, and the mainboard_id,  we need both to make calls.

danielcherubini, 2025-06-23:
Good news ... is this thread made me finally build the better discovery... so now i can have the integration go and FIND the printer for you instead of manually adding it.

cannondale0815, 2025-06-23:
Daniel, I'm so sorry for taking all of your time. It was an educational experience for me for sure. I thank you for spending this time walking me through it!

cannondale0815, 2025-06-23:
Closing this issue. If anyone is having the same issue, have a look at [this post](https://github.com/danielcherubini/elegoo-homeassistant/issues/86#issuecomment-2998209726) for information on what rules may need to be created on your router if you chose to have the router on a subnet that's isolated from your regular LAN.

danielcherubini, 2025-08-07:
@cannondale0815 sorry for not replying sooner. It's been hectic... I also get a CC tomorrow, so i can setup a vlan and test this soon.

But no, nothing has changed on my end

danielcherubini, 2025-08-07:
do you know of a last known working version?