danielcherubini/elegoo-homeassistant #98: Not really a bug but HAOS can't do the proxy thing

suprra, 2025-06-28:
### System Health details

Home Assistant

    Core 2025.3.2
    Supervisor 2025.06.2
    Operating System 14.2
    Frontend 20250306.0

### Checklist

- [x] I have enabled debug logging for my installation.
- [x] I have filled out the issue template to the best of my ability.
- [x] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [x] This issue is not a duplicate issue of any [previous issues](https://github.com/ludeeus/elegoo_printer/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

I have the ha os and it doesn't allow connections that other ha installs would? Sorry not experienced in this, but just putting it here as an informational thing.  Not sure there is a way to get around it.

Thanks for your work, I've been eagerly awaiting the next updates..... no lamp though.... lol xD

### Reproduction steps

1. n/a
2.
3.
...

### Debug logs

```text
n/a
```

danielcherubini, 2025-06-28:
This is a huge help thanks. Seems the upgrade is the thing thats missing... that's not in the spec. Which is a huge thing that this company suffers from, they wrote the spec, then deviated and not updated it. 

https://suchmememanyskill.github.io/OpenCentauri/software/api/

OpenCentauri is tracking the spec as we go.

danielcherubini, 2025-06-28:
Please use this issue for further discussion
https://github.com/danielcherubini/elegoo-homeassistant/issues/54