danielcherubini/elegoo-homeassistant #111: FDM light status sensor

danielcherubini, 2025-06-30:
### Discussed in https://github.com/danielcherubini/elegoo-homeassistant/discussions/104

<div type='discussions-op-text'>

<sup>Originally posted by **bluesk1d** June 30, 2025</sup>
Is it possible to add the internal light as a sensor for FDM printers? They tend to be pretty anemic and people often add supplemental LEDs. It would be great to slave the LEDs to the printer's own light via HA automation.</div>

danielcherubini, 2025-06-30:
@bluesk1d wrote
> Have upgraded to .15 restated home assistant but no light sensor or switch for the light.

try 0.15.2 which just released. You're also going to have to make sure that the "FDM Printer" option is turned on

[image]

bluesk1d, 2025-06-30:
Can confirm that removing and re-adding the printer doesn't resolve this. There isn't anything useful in the logs so I can't be much help (but I'm happy to test or help troubleshoot anything if you need). Just some errors regarding the WIP camera entity. It looks like this would be the first entity that would be an actual switch and not just a sensor. Is it possible that has something to do with it? FWIW a switch would be welcome but for automation purposes, I really just need a sensor if that makes things easier.

`2025-06-30 09:27:38.756 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved (None)
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1538, in _async_process_registry_update_or_remove
    self.async_write_ha_state()
    ~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/components/camera/__init__.py", line 802, in async_write_ha_state
    super().async_write_ha_state()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1019, in async_write_ha_state
    self._async_write_ha_state()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1141, in _async_write_ha_state
    self.__async_calculate_state()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1083, in __async_calculate_state
    if state_attributes := self.state_attributes:
                           ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/camera/__init__.py", line 681, in state_attributes
    attrs = {"access_token": self.access_tokens[-1]}
                             ^^^^^^^^^^^^^^^^^^
AttributeError: 'ElegooMjpegCamera' object has no attribute 'access_tokens'
2025-06-30 09:27:40.438 WARNING (zeroconf-ServiceBrowser-_googlecast._tcp-183) [pychromecast.dial] Failed to determine cast type for host <unknown> (<urlopen error _ssl.c:1011: The handshake operation timed out>) (services:{MDNSServiceInfo(name='Onkyo-TX-NR555-97fe5f9435b7b6ca56f05517860b7d49._googlecast._tcp.local.')})
2025-06-30 09:29:21.701 ERROR (MainThread) [aiohttp.server] Error handling request from 127.0.0.1
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/site-packages/aiohttp/web_protocol.py", line 510, in _handle_request
    resp = await request_handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/aiohttp/web_app.py", line 569, in _handle
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/aiohttp/web_middlewares.py", line 117, in impl
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/security_filter.py", line 92, in security_filter_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/forwarded.py", line 77, in forwarded_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/request_context.py", line 26, in request_context_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/ban.py", line 86, in ban_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/auth.py", line 242, in auth_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/headers.py", line 41, in headers_middleware
    response = await handler(request)
               ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/http.py", line 73, in handle
    result = await handler(request, **request.match_info)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/camera/__init__.py", line 828, in get
    or request.query.get("token") in camera.access_tokens
                                     ^^^^^^^^^^^^^^^^^^^^
AttributeError: 'ElegooMjpegCamera' object has no attribute 'access_tokens'
2025-06-30 09:29:29.111 ERROR (MainThread) [aiohttp.server] Error handling request from 127.0.0.1
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/site-packages/aiohttp/web_protocol.py", line 510, in _handle_request
    resp = await request_handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/aiohttp/web_app.py", line 569, in _handle
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/aiohttp/web_middlewares.py", line 117, in impl
    return await handler(request)
[... 53 lines cut ...]
2025-06-30 09:52:03.004 INFO (Thread-4 (_start_servers_in_thread)) [custom_components.elegoo_printer] Proxy connected to remote printer WebSocket at 192.168.1.254
2025-06-30 09:52:05.410 INFO (MainThread) [custom_components.elegoo_printer] Verified WebSocket connection to Centauri Carbon.
2025-06-30 09:52:05.410 INFO (MainThread) [custom_components.elegoo_printer] Polling Started
[... cut ...]
AttributeError: 'ElegooMjpegCamera' object has no attribute '_webrtc_provider'
2025-06-30 09:52:07.003 DEBUG (MainThread) [custom_components.elegoo_printer] Finished fetching elegoo_printer data in 0.000 seconds (success: True)
2025-06-30 09:52:09.003 DEBUG (MainThread) [custom_components.elegoo_printer] Finished fetching elegoo_printer data in 0.001 seconds (success: True)
2025-06-30 09:52:11.003 DEBUG (MainThread) [custom_components.elegoo_printer] Finished fetching elegoo_printer data in 0.000 seconds (success: True)
`

bluesk1d, 2025-06-30:
I wonder if that is the problem. The light in the printer isn't RGB or dimmable. It's a simple toggle.

bluesk1d, 2025-06-30:
My mistake, didn't realize you meant 2 separate entities. That's interesting since there is definitely no RGB light in the Centauri. Maybe there is a future disco mode planned :)

danielcherubini, 2025-06-30:
[image]

OK they now turn up

danielcherubini, 2025-06-30:
im confident this will work.. im releasing it now

danielcherubini, 2025-06-30:
@bluesk1d try updating to 0.15.5

danielcherubini, 2025-06-30:
@bluesk1d its in the main branch now, but go into light.py, and do both these and test each

1. change the available function to the following
```python
    @property
    def available(self) -> bool:
        """
        Return whether the light entity is currently available.

        If the entity description provides an availability function, uses it with the current light status; otherwise, falls back to the base class availability check.
        """
        return True
```

2. remove the whole available function

Test both and tell me how it goes.

bluesk1d, 2025-06-30:
Both options work with the desired effect for Chamber light.

RGB Light defaults to on and if you toggle it off, the state doesn't stick and it just comes back on. There's no noticeable effect at the actual printer.

bluesk1d, 2025-06-30:
Confirmed working in branch boogaloo :)

bluesk1d, 2025-06-30:
Indeed!
[image]

danielcherubini, 2025-06-30:
Closing this as its fixed!

Next up... removing the need to use the "centauri carbon" option in the selections.. and also to have HA just find and add printers by itself!