danielcherubini/elegoo-homeassistant #133: Can't add a new printer

jtenniswood, 2025-07-05:
### System Health details

[image]
[image]
[image]

### Checklist

- [x] I have enabled debug logging for my installation.
- [x] I have filled out the issue template to the best of my ability.
- [x] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [x] This issue is not a duplicate issue of any [previous issues](https://github.com/ludeeus/elegoo_printer/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

I updated to the latest release (1.2), remove the old printer, added the new one, but it doesn't complete as it says the no_printer_found after it found the printer in the first step

### Reproduction steps

I updated to the latest release (1.2), remove the old printer, added the new one, but it doesn't complete as it says the no_printer_found after it found the printer in the first step

### Debug logs

```text
None
```

jtenniswood, 2025-07-05:
PS Tried with proxy and still the same issue

jtenniswood, 2025-07-05:
PPS I don't understand the 'Enable the proxy server (restart after changing this)', it's unclear what I'm restarting (the printer, home assistant or this integration?)

danielcherubini, 2025-07-05:
> PPS I don't understand the 'Enable the proxy server (restart after changing this)', it's unclear what I'm restarting (the printer, home assistant or this integration?)

it means restart home assistant, but its also that restart isn't needed anymore. 

I'm sorry you're having these issues, first it seems the printer wasn't added correctly, now it seems you're getting one of the 4 no_printer_found errors in the config flow.. im going to add some more to that, so that when it happens again I have more detail on which of the 4 possible scenarios it can be.

But i really would like those debug logs, that would help temendiously.

https://www.home-assistant.io/integrations/logger/

jtenniswood, 2025-07-06:
Here's the logs, let me know if there's something more you need (if it's deeper logging, how to configure that in HA)

[home-assistant_2025-07-06T07-59-37.129Z.log](https://github.com/user-attachments/files/21087277/home-assistant_2025-07-06T07-59-37.129Z.log)

I tried restarting the docker container and re-adding the printer, which almost worked, it was added but nothing worked (no sensors, I'm not sure the controls that were enabled actually work.

[image]

[image]

I can't enable any of these controls

danielcherubini, 2025-07-06:
Ahhh a docker container!!! You probably need to open port 3000, the way the printer contacts is via a call back. See we send a ping out, and it replies, which is where your no printer found error is coming from. 

So ports 3000, but also if you're using the proxy server 3030, and 3031

danielcherubini, 2025-07-06:
Im not sure its a port problem...

looking at the logs..

```
2025-07-06 08:56:03.261 INFO (SyncWorker_2) [custom_components.elegoo_printer] Broadcasting for printer/proxy discovery...
2025-07-06 08:56:03.383 INFO (SyncWorker_2) [custom_components.elegoo_printer] Discovery response received from ('192.168.1.80', 3000)
2025-07-06 08:56:03.383 INFO (SyncWorker_2) [custom_components.elegoo_printer] Discovered: Centauri Carbon (192.168.1.80)
2025-07-06 08:56:04.515 DEBUG (MainThread) [homeassistant.components.google_assistant.report_state] Scheduling report state for light.jessicas_bed_light: {'online': True, 'on': False}
2025-07-06 08:56:05.593 ERROR (MainThread) [homeassistant.components.google_assistant.http] Request for https://homegraph.googleapis.com/v1/devices:reportStateAndNotification failed: 400
2025-07-06 08:56:11.668 INFO (MainThread) [custom_components.elegoo_printer] Client connecting to WebSocket at: ws://192.168.1.80:3030/websocket
2025-07-06 08:56:11.668 INFO (MainThread) [custom_components.elegoo_printer] Client successfully connected via proxy to: Centauri Carbon
2025-07-06 08:56:12.174 INFO (Thread-15 (run_forever)) [websocket] Websocket connected
2025-07-06 08:56:14.175 INFO (MainThread) [custom_components.elegoo_printer] Verified WebSocket connection to Centauri Carbon.
2025-07-06 08:56:14.175 INFO (MainThread) [homeassistant.setup] Setting up elegoo_printer
2025-07-06 08:56:14.176 INFO (MainThread) [custom_components.elegoo_printer] Client connecting to WebSocket at: ws://192.168.1.80:3030/websocket
2025-07-06 08:56:14.176 INFO (MainThread) [custom_components.elegoo_printer] Client successfully connected via proxy to: Centauri Carbon
2025-07-06 08:56:14.745 INFO (Thread-16 (run_forever)) [websocket] Websocket connected
2025-07-06 08:56:16.746 INFO (MainThread) [custom_components.elegoo_printer] Verified WebSocket connection to Centauri Carbon.
2025-07-06 08:56:16.746 INFO (MainThread) [custom_components.elegoo_printer] Polling Started
2025-07-06 08:56:18.767 INFO (MainThread) [homeassistant.components.sensor] Setting up elegoo_printer.sensor
2025-07-06 08:56:18.769 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.elegoo_printer entity: sensor.centauri_carbon_current_print_time
2025-07-06 08:56:18.770 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.elegoo_printer entity: sensor.centauri_carbon_remaining_print_time
2025-07-06 08:56:18.771 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.elegoo_printer entity: sensor.centauri_carbon_total_layers
2025-07-06 08:56:18.771 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.elegoo_printer entity: sensor.centauri_carbon_current_layer
2025-07-06 08:56:18.772 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.elegoo_printer entity: sensor.centauri_carbon_remaining_layers
2025-07-06 08:56:18.772 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.elegoo_printer entity: sensor.centauri_carbon_percent_complete
2025-07-06 08:56:18.773 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.elegoo_printer entity: sensor.centauri_carbon_file_name
2025-07-06 08:56:18.773 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.elegoo_printer entity: sensor.centauri_carbon_current_status
2025-07-06 08:56:18.773 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.elegoo_printer entity: sensor.centauri_carbon_print_status
2025-07-06 08:56:18.774 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.elegoo_printer entity: sensor.centauri_carbon_print_error
2025-07-06 08:56:18.774 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.elegoo_printer entity: sensor.centauri_carbon_error_status_reason
2025-07-06 08:56:18.775 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.elegoo_printer entity: sensor.centauri_carbon_video_stream_connected
2025-07-06 08:56:18.776 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.elegoo_printer entity: sensor.centauri_carbon_video_stream_max
2025-07-06 08:56:18.776 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.elegoo_printer entity: sensor.centauri_carbon_box_temp
2025-07-06 08:56:18.777 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.elegoo_printer entity: sensor.centauri_carbon_box_target_temp
2025-07-06 08:56:18.777 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.elegoo_printer entity: sensor.centauri_carbon_nozzle_temperature
2025-07-06 08:56:18.778 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.elegoo_printer entity: sensor.centauri_carbon_bed_temperature
2025-07-06 08:56:18.778 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.elegoo_printer entity: sensor.centauri_carbon_z_offset
2025-07-06 08:56:18.779 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.elegoo_printer entity: sensor.centauri_carbon_model_fan_speed
2025-07-06 08:56:18.779 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.elegoo_printer entity: sensor.centauri_carbon_auxiliary_fan_speed
[... cut ...]
2025-07-06 08:57:12.687 ERROR (Thread-15 (run_forever)) [custom_components.elegoo_printer] Connection to Centauri Carbon (via proxy) error: Connection to remote host was lost.
2025-07-06 08:57:12.687 INFO (Thread-15 (run_forever)) [websocket] Connection to remote host was lost. - reconnect
2025-07-06 08:57:12.949 INFO (Thread-15 (run_forever)) [websocket] reconnect() - retrying in 1 seconds [5 frames in stack]
2025-07-06 08:57:13.957 INFO (Thread-15 (run_forever)) [websocket] Websocket connected
```

Then the websocket keeps having problems. When was the last time you power cycled your printer? Im wondering if you have hit the max connections or something...

danielcherubini, 2025-07-06:
To explain the max connections, which also explains the need for the proxy... 

The printer can have 4 maximum connections into it.. and video counts towards that 4, so 1 websocket, and 1 video is half your connections, So your slicer takes half.. and it doesn't like giving those connections up.

You open the webui? that also takes half your connections.. now you hit the max. 

So the proxy works as a kind of way to help fix that issue, so you can connect your slicers/etc though the proxy. Taking only 2 connections for as many things as you can

jtenniswood, 2025-07-06:
The printer has only been switched on for about 30mins. I didn't have these issues before v1 build.
I tried with the proxy but it had the same results, all the sensors/controls are disabled or unavailable.

[home-assistant_2025-07-06T08-51-46.708Z.log](https://github.com/user-attachments/files/21087466/home-assistant_2025-07-06T08-51-46.708Z.log)

Personally I don't want/need the camera setup, because I'm using cloudflare, I can't stream the camera via home assistant (some cross site config issue I have never figured out, streams just don't load, so I don't use them).

jtenniswood, 2025-07-06:
I can't enable any of the sensors

[image]

danielcherubini, 2025-07-06:
https://www.reddit.com/r/homeassistant/comments/1f8ifa8/the_device_is_disabled_by_config_entry/

Can you try that? There's a screenshot lower in the replies

I don't know why you're the only one with this specific issue. The integration doesn't disable entities, I never set that up. They're either added or not.

jtenniswood, 2025-07-06:
Amazing that fixed it. Bonkers that it was able to get into that state!

closed 2025-07-06 by jtenniswood

danielcherubini, 2025-07-06:
I had a feeling that something was weird here. Thanks for all your debugging. Have a great week!