danielcherubini/elegoo-homeassistant #140: High CPU Load

danielcherubini, 2025-07-08:
This probably belongs in a ticket but I've not really done a full debug so can't in good concience tick all the boxes :)

I added this integration recently and everything looked good but then I turned my S4U off and after about 24 hours the CPU load used by the Homeassistant container was up over 200%.  I rebooted the server, and again about 24 hours later the load was back up over 200% ... I tried to do some diagnostics but unfortunately I only got a partial run of py-spy before the container just died and restarted, but what I got seemed to indicate it was spending all its time in a websocket call.

My system had been running fine prior to adding this and a satellite to wyonming, so I removed both of these and the problem did not occur, so I added the wyoming satellite back in and it still hasn't reoccured, so by process of elimination and the websocket call being where home assistant was spending its time I'm thinking it is to do with this integration.

is it possible that when the device isn't present the retries to establish a connection are not closing their websockets correctly, leading to the CPU spike ?

_Originally posted by @LucidityCrash in https://github.com/danielcherubini/elegoo-homeassistant/discussions/139_

danielcherubini, 2025-07-08:
Leaving this open till tomorrow morning when if my cpu load isnt high, its cool

danielcherubini, 2025-07-08:
I'm going to make a change to how disconnect is called. If your able to run an update in the next hour and check. That would be cool

danielcherubini, 2025-07-08:
@LucidityCrash

https://github.com/danielcherubini/elegoo-homeassistant/releases/tag/v1.3.4

LucidityCrash, 2025-07-08:
Just done that :) 

[image]

danielcherubini, 2025-07-08:
is it this?

https://github.com/rhasspy/wyoming-satellite

Id like to know more about this other integration

LucidityCrash, 2025-07-08:
> is it this?
> 
> https://github.com/rhasspy/wyoming-satellite
> 
> Id like to know more about this other integration

yep thats the one.

LucidityCrash, 2025-07-08:
and still the issue remains ... this is nuts ... enable this integration with my printer turned off and the wyoming satellite disconnects

[image]

danielcherubini, 2025-07-08:
@LucidityCrash i had a thought, what if its NOT the disconnect at all.... but the timeout!

```
2025-07-08 18:07:26.541 DEBUG (MainThread) [homeassistant.components.wyoming.assist_satellite] Disconnecting from satellite
2025-07-08 18:07:26.541 DEBUG (MainThread) [homeassistant.components.wyoming.assist_satellite] Connecting to satellite at 192.168.78.38:10700
2025-07-08 18:07:26.594 DEBUG (MainThread) [homeassistant.components.wyoming.assist_satellite] Connected to satellite
2025-07-08 18:07:26.614 DEBUG (MainThread) [homeassistant.components.wyoming.assist_satellite] Updated client info: Info(asr=[], tts=[], handle=[], intent=[], wake=[WakeProgram(name='openwakeword', attribution=Attribution(name='dscripka', url='https://github.com/dscripka/openWakeWord'), installed=True, description='An open-source audio wake word (or phrase) detection framework with a focus on performance and simplicity.', version='1.10.1', models=[WakeModel(name='hey_rhasspy_v0.1', attribution=Attribution(name='dscripka', url='https://github.com/dscripka/openWakeWord'), installed=True, description='hey rhasspy', version='v0.1', languages=[], phrase='hey rhasspy'), WakeModel(name='alexa_v0.1', attribution=Attribution(name='dscripka', url='https://github.com/dscripka/openWakeWord'), installed=True, description='alexa', version='v0.1', languages=[], phrase='alexa'), WakeModel(name='ok_nabu_v0.1', attribution=Attribution(name='dscripka', url='https://github.com/dscripka/openWakeWord'), installed=True, description='ok nabu', version='v0.1', languages=[], phrase='ok nabu'), WakeModel(name='hey_mycroft_v0.1', attribution=Attribution(name='dscripka', url='https://github.com/dscripka/openWakeWord'), installed=True, description='hey mycroft', version='v0.1', languages=[], phrase='hey mycroft'), WakeModel(name='hey_jarvis_v0.1', attribution=Attribution(name='dscripka', url='https://github.com/dscripka/openWakeWord'), installed=True, description='hey jarvis', version='v0.1', languages=[], phrase='hey jarvis'), WakeModel(name='glados', attribution=Attribution(name='dscripka', url='https://github.com/dscripka/openWakeWord'), installed=True, description='glados', version=None, languages=[], phrase='glados'), WakeModel(name='hey_glados', attribution=Attribution(name='dscripka', url='https://github.com/dscripka/openWakeWord'), installed=True, description='hey glados', version=None, languages=[], phrase='hey glados')])], mic=[], snd=[], satellite=Satellite(name='GLaDOS-1', attribution=Attribution(name='', url=''), installed=True, description='GLaDOS-1', version='1.4.1', area=None, has_vad=None, active_wake_words=None, max_active_wake_words=None, supports_trigger=None))
2025-07-08 18:08:43.637 DEBUG (MainThread) [custom_components.elegoo_printer] CONFIGURATION {'brand': 'ELEGOO', 'connection': 'f25273b12b094c5a8b9513a30ca60049', 'firmware': 'V1.2.7', 'id': '48412c250d160100', 'ip_address': '192.168.1.246', 'model': 'Saturn 4 Ultra', 'name': '3D Printer', 'printer_type': 'resin', 'protocol': 'V3.0.0', 'proxy_enabled': False}
2025-07-08 18:08:43.637 DEBUG (MainThread) [custom_components.elegoo_printer] Connecting to printer: 3D Printer at 192.168.1.246 with proxy enabled False and printer.proxy: False
2025-07-08 18:08:43.637 DEBUG (MainThread) [custom_components.elegoo_printer] Connecting to printer: 3D Printer at 192.168.1.246 proxy_enabled: False
2025-07-08 18:08:43.637 INFO (MainThread) [custom_components.elegoo_printer] Client connecting to WebSocket at: ws://192.168.1.246:3030/websocket
2025-07-08 18:08:46.754 DEBUG (Thread-78 (run_forever)) [custom_components.elegoo_printer] Connection to 3D Printer error: [Errno 113] Host is unreachable
2025-07-08 18:08:46.754 DEBUG (Thread-78 (run_forever)) [custom_components.elegoo_printer] Connection to 3D Printer closed: None (None)
2025-07-08 18:08:48.638 WARNING (MainThread) [custom_components.elegoo_printer] Failed to connect WebSocket to 3D Printer within timeout.
2025-07-08 18:08:48.638 INFO (MainThread) [custom_components.elegoo_printer] Closing connection to printer
2025-07-08 18:08:48.690 DEBUG (MainThread) [homeassistant.components.wyoming.assist_satellite] TimeoutError: 
2025-07-08 18:08:48.691 WARNING (MainThread) [homeassistant.components.wyoming.assist_satellite] Satellite has been disconnected. Reconnecting in 10 second(s)
```

Look at the second last line.... "TimeoutError"

danielcherubini, 2025-07-08:
Reason i ask, is that i set `websocket.setDefaultTimeout(5)` when connecting to the websocket on the printer

My thinking here... is ... what if that default timeout is shared across anything that uses websockets?

danielcherubini, 2025-07-08:
[image]

danielcherubini, 2025-07-08:
HOLY MOLY!

LucidityCrash, 2025-07-08:
It would also explain the increase in the number of timeouts to my Wiser Central heating controller ... It can be a bit flaky at times so I didn't really pay it much attention, but there were definitely more timeouts.

danielcherubini, 2025-07-08:
OK, try 1.3.6

danielcherubini, 2025-07-08:
No. It's good. I like solving hard puzzles. 

Does it do this if the printer is on?

If not, then I think I might have to remove that disconnect again and try that.

danielcherubini, 2025-07-08:
Ok. Looking at the other repo. Looks like you're running that websocket server on the same machine that home assistant is on right?

Using the script? 

https://github.com/rhasspy/wyoming-satellite/blob/master/examples%2Fwebsocket-service%2Fserver.py

Tell me more about the system it runs on. You use home assistant OS? Or I'm going to assume something like a RPi or a VM. But not HAOS, but instead home assistant core.

LucidityCrash, 2025-07-08:
No ... it runs on a rpi zero 2 that is just running raspian ... No HA involved at all.

danielcherubini, 2025-07-08:
Right so you have two devices. One server which is your speaker/microphone thing. Thats on aRPi

Then home assistant is on something else, and this is running in the background on the home assistant device connecting to the remote?

LucidityCrash, 2025-07-08:
what do you mean by "this" - the wyoming-satellite ... all of that runs on the seperate device.  HA connects to the remote device on port 10700 and uses a protocol called wyoming.

I disabled debug and then turned on my S4U ... the satellite disconnects stop : 

[home-assistant_2025-07-08T19-59-02.031Z.log](https://github.com/user-attachments/files/21130562/home-assistant_2025-07-08T19-59-02.031Z.log)

danielcherubini, 2025-07-08:
Sorry by this I meant that other repository.. so ok all that runs on a separate device..

How does the websockets come into play?

It sounds like all that is being run on a seperate device to home assistant

danielcherubini, 2025-07-08:
RIGHT... i see
https://github.com/home-assistant/core/blob/70c01efe570c3dbdec4952e5da5b529c465a5e1b/homeassistant/components/wyoming/websocket_api.py

danielcherubini, 2025-07-08:
[image]

danielcherubini, 2025-07-08:
ok @LucidityCrash i think its ready.

danielcherubini, 2025-07-08:
i think i might make this a pre-release, meaning you will need to manually download it in the HACS menu, by selecting re-download

then selecting v1.4.0b1

danielcherubini, 2025-07-08:
[image]

LucidityCrash, 2025-07-09:
Sorry for the delay, apparently unlike you, I need my beauty sleep :) 

[image]

After reboot and enabling the integration with the printer off.  Previoustly for each of those there would have been a corresponding Satellite disconnect message.  Connects automatically when the printer is turned on, looks like there is some kind of exponential backoff in the retries.  Then turned the printer off.

Oh that is interesting : 

[image]

the sensors retain their last known value.

LucidityCrash, 2025-07-09:
CPU load looks pretty normal after having my printer switched off all day.

danielcherubini, 2025-07-09:
well, the original issue was high cpu load, the secondary was the websocket disconnect... so i think both are solved.

commit 3250e240 (2025-07-08): Fixes #140 (#141)

* Fixes #140

* ruff

* moving debug py

* nitpick

* fixing lint
closed 2025-07-09 by danielcherubini