danielcherubini/elegoo-homeassistant #157: Camera Fetching Errors

inventor7777, 2025-07-13:
### System Health details

Update in just a sec

### Checklist

- [x] I have enabled debug logging for my installation.
- [x] I have filled out the issue template to the best of my ability.
- [x] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [x] This issue is not a duplicate issue of any [previous issues](https://github.com/ludeeus/elegoo_printer/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

Camera errors come up in logs even with only HA and slicer connected

### Reproduction steps

1.
2.
3.
...

### Debug logs

```text
2025-07-12 18:51:11.305 ERROR (MainThread) [homeassistant.components.mjpeg] Error getting new camera image from Centauri Carbon Camera: All connection attempts failed

I also see this

2025-07-12 18:54:33.535 ERROR (MainThread) [custom_components.elegoo_printer] Error fetching elegoo_printer data: Elegoo Printer is Offline
2025-07-12 18:55:43.536 ERROR (MainThread) [custom_components.elegoo_printer] WebSocket listener exception: WebSocket connection error: No PONG received after 10.0 seconds
2025-07-12 18:56:16.537 ERROR (MainThread) [custom_components.elegoo_printer] WebSocket listener exception: WebSocket connection error: No PONG received after 10.0 seconds
2025-07-12 18:57:10.535 ERROR (MainThread) [custom_components.elegoo_printer] WebSocket listener exception: WebSocket connection error: No PONG received after 10.0 seconds
```

### Diagnostics dump

See above

danielcherubini, 2025-07-13:
@inventor7777 this looks like it might be a bug in the proxy server

inventor7777, 2025-07-13:
Does my proxy server need to be turned on? I turned it off because it caused everything else to stop working. I updated to the beta however and now I get this

[image]

danielcherubini, 2025-07-13:
No. But. That log is interesting. 

Can you get me the debug logs please

danielcherubini, 2025-07-13:
Proxy server on or off doesn't matter for this. 

And. If nothing is working when it's on, then it's probably best to restart the printer, which resets the maximum count of active devices, and then restart home assistant. 

But let's just focus on getting the camera up again.

inventor7777, 2025-07-13:
@danielcherubini Today I’m printing an 18hr print so I moved it out of my bedroom temporarily. 

Anyway, it changed the IP address down 1 (different mesh node) and ElegooPrinter won’t let me update the IP for some reason.

The logs have the old IP address still which is bizarre. The old address ends in .123 and the temp one ends in .122 yet the logs aren’t updating, so that could be the issue.

Here are some images

[image]
[image]

inventor7777, 2025-07-13:
If you are wondering why the first image has the old ip it’s because it resets the field back to original when the error appears (I type the IP ending in .122 and it resets it back to the ip ending in .123

danielcherubini, 2025-07-13:
Ok that's a new bug.

danielcherubini, 2025-07-13:
For the time being. You can remove it and re-add it. It keeps all your automations and stuff working fine.

danielcherubini, 2025-07-13:
Proxy on or off?

danielcherubini, 2025-07-13:
I believe you need to click the camera entity, and try to view it.

When you try and view it, it checks if the printer has enough connections left and then sends the url to home assistant.

Which is what im trying to find out in the debug logs :D

inventor7777, 2025-07-13:
@danielcherubini 

This is what I got

2025-07-13 17:47:30.684 DEBUG (MainThread) [custom_components.elegoo_printer.elegoo_sdcp] get_printer_current_task task_id: db790de5-3973-475e-9dba-f9ed22af7283
2025-07-13 17:47:30.685 DEBUG (MainThread) [custom_components.elegoo_printer.elegoo_sdcp] get_printer_current_task: got cached task
2025-07-13 17:47:30.685 DEBUG (MainThread) [custom_components.elegoo_printer] Finished fetching elegoo_printer data in 2.003 seconds (success: True)
2025-07-13 17:47:30.966 DEBUG (MainThread) [custom_components.elegoo_printer] Sending printer video: {'status': <ElegooVideoStatus.SUCCESS: 0>, 'video_url': '192.168.50.122:3031/video'}
2025-07-13 17:47:30.966 DEBUG (MainThread) [custom_components.elegoo_printer] stream_source: Video is OK, getting stream source
2025-07-13 17:47:30.966 DEBUG (MainThread) [custom_components.elegoo_printer] stream_source: Proxy is disabled using printer video url: 192.168.50.122:3031/video
2025-07-13 17:47:30.968 ERROR (MainThread) [homeassistant.components.mjpeg] Error getting new camera image from Centauri Carbon Camera: Request URL is missing an 'http://' or 'https://' protocol.
2025-07-13 17:47:31.053 DEBUG (MainThread) [custom_components.elegoo_printer.elegoo_sdcp] get_printer_current_task task_id: db790de5-3973-475e-9dba-f9ed22af7283
2025-07-13 17:47:31.053 DEBUG (MainThread) [custom_components.elegoo_printer.elegoo_sdcp] get_printer_current_task: got cached task
2025-07-13 17:47:34.684 DEBUG (MainThread) [custom_components.elegoo_printer.elegoo_sdcp] get_printer_current_task task_id: db790de5-3973-475e-9dba-f9ed22af7283
2025-07-13 17:47:34.685 DEBUG (MainThread) [custom_components.elegoo_printer.elegoo_sdcp] get_printer_current_task: got cached task
2025-07-13 17:47:34.685 DEBUG (MainThread) [custom_components.elegoo_printer] Finished fetching elegoo_printer data in 2.002 seconds (success: True)
2025-07-13 17:47:38.685 DEBUG (MainThread) [custom_components.elegoo_printer.elegoo_sdcp] get_printer_current_task task_id: db790de5-3973-475e-9dba-f9ed22af7283
2025-07-13 17:47:38.686 DEBUG (MainThread) [custom_components.elegoo_printer.elegoo_sdcp] get_printer_current_task: got cached task
2025-07-13 17:47:38.686 DEBUG (MainThread) [custom_components.elegoo_printer] Finished fetching elegoo_printer data in 2.003 seconds (success: True)
2025-07-13 17:47:42.684 DEBUG (MainThread) [custom_components.elegoo_printer.elegoo_sdcp] get_printer_current_task task_id: db790de5-3973-475e-9dba-f9ed22af7283
2025-07-13 17:47:42.685 DEBUG (MainThread) [custom_components.elegoo_printer.elegoo_sdcp] get_printer_current_task: got cached task
2025-07-13 17:47:42.685 DEBUG (MainThread) [custom_components.elegoo_printer] Finished fetching elegoo_printer data in 2.002 seconds (success: True)
2025-07-13 17:47:46.684 DEBUG (MainThread) [custom_components.elegoo_printer.elegoo_sdcp] get_printer_current_task task_id: db790de5-3973-475e-9dba-f9ed22af7283
2025-07-13 17:47:46.684 DEBUG (MainThread) [custom_components.elegoo_printer.elegoo_sdcp] get_printer_current_task: got cached task
2025-07-13 17:47:46.685 DEBUG (MainThread) [custom_components.elegoo_printer] Finished fetching elegoo_printer data in 2.002 seconds (success: True)
2025-07-13 17:47:50.684 DEBUG (MainThread) [custom_components.elegoo_printer.elegoo_sdcp] get_printer_current_task task_id: db790de5-3973-475e-9dba-f9ed22af7283
2025-07-13 17:47:50.684 DEBUG (MainThread) [custom_components.elegoo_printer.elegoo_sdcp] get_printer_current_task: got cached task
2025-07-13 17:47:50.684 DEBUG (MainThread) [custom_components.elegoo_printer] Finished fetching elegoo_printer data in 2.002 seconds (success: True)
2025-07-13 17:47:54.684 DEBUG (MainThread) [custom_components.elegoo_printer.elegoo_sdcp] get_printer_current_task task_id: db790de5-3973-475e-9dba-f9ed22af7283
2025-07-13 17:47:54.685 DEBUG (MainThread) [custom_components.elegoo_printer.elegoo_sdcp] get_printer_current_task: got cached task
2025-07-13 17:47:54.685 DEBUG (MainThread) [custom_components.elegoo_printer] Finished fetching elegoo_printer data in 2.003 seconds (success: True)
2025-07-13 17:47:58.685 DEBUG (MainThread) [custom_components.elegoo_printer.elegoo_sdcp] get_printer_current_task task_id: db790de5-3973-475e-9dba-f9ed22af7283
2025-07-13 17:47:58.685 DEBUG (MainThread) [custom_components.elegoo_printer.elegoo_sdcp] get_printer_current_task: got cached task
2025-07-13 17:47:58.685 DEBUG (MainThread) [custom_components.elegoo_printer] Finished fetching elegoo_printer data in 2.002 seconds (success: True)
2025-07-13 17:48:02.684 DEBUG (MainThread) [custom_components.elegoo_printer.elegoo_sdcp] get_printer_current_task task_id: db790de5-3973-475e-9dba-f9ed22af7283
2025-07-13 17:48:02.684 DEBUG (MainThread) [custom_components.elegoo_printer.elegoo_sdcp] get_printer_current_task: got cached task
2025-07-13 17:48:02.684 DEBUG (MainThread) [custom_components.elegoo_printer] Finished fetching elegoo_printer data in 2.002 seconds (success: True)
2025-07-13 17:48:06.684 DEBUG (MainThread) [custom_components.elegoo_printer.elegoo_sdcp] get_printer_current_task task_id: db790de5-3973-475e-9dba-f9ed22af7283
2025-07-13 17:48:06.685 DEBUG (MainThread) [custom_components.elegoo_printer.elegoo_sdcp] get_printer_current_task: got cached task
2025-07-13 17:48:06.685 DEBUG (MainThread) [custom_components.elegoo_printer] Finished fetching elegoo_printer data in 2.003 seconds (success: True)
2025-07-13 17:48:10.684 DEBUG (MainThread) [custom_components.elegoo_printer.elegoo_sdcp] get_printer_current_task task_id: db790de5-3973-475e-9dba-f9ed22af7283
2025-07-13 17:48:10.684 DEBUG (MainThread) [custom_components.elegoo_printer.elegoo_sdcp] get_printer_current_task: got cached task

danielcherubini, 2025-07-13:
GOT IT!

```
2025-07-13 17:47:30.966 DEBUG (MainThread) [custom_components.elegoo_printer] Sending printer video: {'status': <ElegooVideoStatus.SUCCESS: 0>, 'video_url': '192.168.50.122:3031/video'}
``` 

For some reason your printer is now NOT sending the http part! super easy fix!

danielcherubini, 2025-07-13:
https://github.com/danielcherubini/elegoo-homeassistant/releases/tag/v1.4.7b2

Took me more time to get this release out than to write the fix!

danielcherubini, 2025-07-13:
Thats your automation, something changed in it, probably when you removed and re-added the printer, the name of the sensor changed.. its an easy fix but something you need to do. Just open the automation and update the triggers/actions