danielcherubini/elegoo-homeassistant #160: Multiple printers not creating properly

cdunn95, 2025-07-13:
### System Health details

## System Information

version | core-2025.3.4
-- | --
installation_type | Home Assistant Container
dev | false
hassio | false
docker | true
user | root
virtualenv | false
python_version | 3.13.2
os_name | Linux
os_version | 6.8.0-62-generic
arch | x86_64
timezone | America/New_York
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
HACS Data | ok
GitHub API Calls Remaining | 4982
Installed Version | 2.0.5
Stage | startup
Available Repositories | 2102
Downloaded Repositories | 14

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Dashboards</summary>

dashboards | 4
-- | --
resources | 2
views | 1
mode | storage

</details>

<details><summary>Network Configuration</summary>

adapters | lo (disabled), enp2s0 (enabled, default, auto), docker0 (disabled), vethc67d3e5 (disabled)
-- | --
ipv4_addresses | lo (127.0.0.1/8), enp2s0 (192.168.1.100/24), docker0 (172.17.0.1/16), vethc67d3e5 ()
ipv6_addresses | lo (::1/128), enp2s0 (fd6f:949e:2c2e:d66e:e251:d8ff:fe18:787a/64, fe80::e251:d8ff:fe18:787a/64), docker0 (fe80::64c3:75ff:fec0:f7a4/64), vethc67d3e5 (fe80::45b:7eff:feff:a3e4/64)
announce_addresses | 192.168.1.100, fd6f:949e:2c2e:d66e:e251:d8ff:fe18:787a, fe80::e251:d8ff:fe18:787a

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | June 29, 2025 at 3:39 PM
-- | --
current_recorder_run | July 13, 2025 at 1:42 AM
estimated_db_size | 224.00 MiB
database_engine | sqlite
database_version | 3.48.0

</details>

### Checklist

- [x] I have enabled debug logging for my installation.
- [x] I have filled out the issue template to the best of my ability.
- [x] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [x] This issue is not a duplicate issue of any [previous issues](https://github.com/ludeeus/elegoo_printer/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

Adding multiple Centauri Carbons results in only one being added successfully, with the subsequent device containing only one entity

### Reproduction steps

1. Add multiple Centauri Carbons
2. Check integration page for devices

### Debug logs

```text
Attached to diagnostics dump (debug logs does not have a section to upload files)
```

### Diagnostics dump

[error_log-5.txt](https://github.com/user-attachments/files/21202742/error_log-5.txt)

danielcherubini, 2025-07-13:
Hey .. sorry for not looking at this sooner, thanks for making an issue...

danielcherubini, 2025-07-13:
I'll have to tackle that sub issue first, then i can tackle this.

cdunn95, 2025-07-14:
> Hey .. sorry for not looking at this sooner, thanks for making an issue...
> 
> 

Not a problem, sorry for not creating the issue ticket sooner! Hope you can get it fixed soon!

danielcherubini, 2025-07-21:
https://github.com/danielcherubini/elegoo-homeassistant/releases/tag/v1.4.9b1

This is ready for testing

cdunn95, 2025-08-12:
@danielcherubini Can we get this branch merged in, please? I haven't had any issues with multi printer running 1.4.9b1 (and I've been running since its release)

danielcherubini, 2025-08-12:
This is fixed? I'm confused on why you closed it

danielcherubini, 2025-08-12:
That's great it's working for you. I noticed some migration problems going from the previous version to this one. So I'll deal with that next. But thanks for confirming that the fix worked.

cdunn95, 2025-08-12:
> This is fixed? I'm confused on why you closed it

It is fixed in [1.4.9b1](https://github.com/danielcherubini/elegoo-homeassistant/releases/tag/v1.4.9b1), but the patch included in that has yet to be merged into main and built. I see that you have #162 still open, but the latest commits to it have been surrounding _removing_ the entities, not the _creation_ of them, which is what this issue was originally intended for. 

I would ask that you branch create a new branch from d416c31ca63602da36462369fefba7b21f0fd3a5 that is focused on fixing the removal of entities, revert bugfix/more_unique_names to 58f7355a1221f6def8424562aa1188c3875e8cdb, and go through with PR on that branch so that the next release includes the capability to have multiple printers, even if they are not currently able to leverage the proxy features. 

Despite me feeling like the original intention of this ticket has been resolved in that pre-release build, I feel that I may have overstepped by closing the issue; as such, I am reopening. My apologies.

danielcherubini, 2025-08-12:
> It is fixed in [1.4.9b1](https://github.com/danielcherubini/elegoo-homeassistant/releases/tag/v1.4.9b1), but the patch included in that has yet to be merged into main and built. I see that you have [...]

I understand you want this asap. But I need to solve the issue that's there first. 

Basically the migration which is the "removing" you're talking about is about one or two entities dangle when the migration completes. Then home assistant has two devices instead of one. One with everything and another that's either empty or with a single dead entity. 

The solutions is easy. I'm going to go nuclear. Remove the device and re-add it as the migration. 

I can't just skip this otherwise I'm going to have a lot of issues and questions about why there's these extra things.

commit 99a8487a (2025-08-14): Uses MachineID as unique ID to support multiple printers. (#162)

* Fixes #160

* 1.4.7b1

* 1.4.8b1

* adds migration

* v1.4.8b2

* lint fix

* 1.4.8b3

* ok this should do it

* updates the update for ids

* Apply suggestion from @coderabbitai[bot]

* 1.4.8b4

* v1.4.9b1

* moving to 1.5.0b1

* better removal of entities

* going nuclear

* fixes

* bumping version to beta

---------
closed 2025-08-14 by danielcherubini