danielcherubini/elegoo-homeassistant #165: Timelapse recording file fails to download

bystand3r, 2025-07-21:
### System Health details

When connected via the proxy I am unable to download the videos. The web interface loads and runs until the export button is clicked, the page times out at the video link file. If i update the link in the address bar to the printers ip address it will let me view that specific recording  and download. I can’t download when using the proxy ip. 

### Checklist

- [x] I have enabled debug logging for my installation.
- [x] I have filled out the issue template to the best of my ability.
- [x] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [x] This issue is not a duplicate issue of any [previous issues](https://github.com/ludeeus/elegoo_printer/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

When connected via the proxy I am unable to download the videos. The web interface loads and runs until the export button is clicked, the page times out at the video link file. If i update the link in the address bar to the printers ip address it will let me view that specific recording  and download. I can’t download when using the proxy ip. 

### Reproduction steps

1. Connect printer to proxy 
2. Browse to web interface on pc or phone
3. Select a video file and click export 
4. page goes  through stages as if it’s going to download then times-out
5. address in bar shows the printer and the reference into where the exact file is
6. if we edit this address by changing the ip to the actual it then reloads the page and shows the video on a webpage and can be downloaded.
...

### Debug logs

```text
Can’t get these sorry
```

bystand3r, 2025-07-25:
I will certainly do whatever I need to for assisting with this. Overseas at the moment in holiday and whilst I can connect to the printer the web interface doesn't load the file export section when connecting from an iPhone :(

danielcherubini, 2025-07-25:
Hahah. I'm also overseas on holidays. 

Have been the whole month. 

Take your time. No rush. Enjoy the vacation

danielcherubini, 2025-08-08:
Found the issue

the elegoo webpage sends you to a different url than the proxy is running on

http://PRINTERIP:80/local/aic_tlp/file.mp4

the problem is port 80

closed 2025-09-09 by danielcherubini