danielcherubini/elegoo-homeassistant #171: When printer is turned off, logs are spammed and it constantly retries setup.

inventor7777, 2025-08-03:
### Checklist

- [x] I have filled out the template to the best of my ability.
- [x] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [x] This issue is not a duplicate feature request of [previous feature requests](https://github.com/ludeeus/elegoo_printer/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

When I turn my printer's switch off, the integration goes to "Failed setup, will retry" and it spams the logs with error messages.

### Describe the solution you'd like

I think instead of failing outright and logging errors, it should just make all of the sensors go to "Unavailable" like a lot of other integrations do.

### Describe alternatives you've considered

An alternative could be having a switch to make the integration active, so I could link it to my printer's smart plug to only be polling the printer when it's on. 

### Additional context

Since the entire integration is built on a connection to the printer, this could be quite a complex feature request. No hurry and if you don't think it's worth it that's okay. I don't believe there's a way to turn off an integration in a Home Assistant script, but if there is, I can do that instead.

<details>
  <summary>Annoying Logs</summary>
2025-08-03 00:15:08.237 WARNING (MainThread) [custom_components.elegoo_printer] Failed to connect WebSocket to Centauri Carbon: Cannot connect to host 192.168.50.122:3030 ssl:default [Connect call failed ('192.168.50.122', 3030)]
2025-08-03 00:16:31.757 WARNING (MainThread) [custom_components.elegoo_printer] Failed to connect WebSocket to Centauri Carbon: Cannot connect to host 192.168.50.122:3030 ssl:default [Connect call failed ('192.168.50.122', 3030)]
2025-08-03 00:17:55.278 WARNING (MainThread) [custom_components.elegoo_printer] Failed to connect WebSocket to Centauri Carbon: Cannot connect to host 192.168.50.122:3030 ssl:default [Connect call failed ('192.168.50.122', 3030)]
2025-08-03 00:19:18.637 WARNING (MainThread) [custom_components.elegoo_printer] Failed to connect WebSocket to Centauri Carbon: Cannot connect to host 192.168.50.122:3030 ssl:default [Connect call failed ('192.168.50.122', 3030)]
2025-08-03 00:20:42.237 WARNING (MainThread) [custom_components.elegoo_printer] Failed to connect WebSocket to Centauri Carbon: Cannot connect to host 192.168.50.122:3030 ssl:default [Connect call failed ('192.168.50.122', 3030)]
2025-08-03 00:22:05.677 WARNING (MainThread) [custom_components.elegoo_printer] Failed to connect WebSocket to Centauri Carbon: Cannot connect to host 192.168.50.122:3030 ssl:default [Connect call failed ('192.168.50.122', 3030)]
2025-08-03 00:23:28.877 WARNING (MainThread) [custom_components.elegoo_printer] Failed to connect WebSocket to Centauri Carbon: Cannot connect to host 192.168.50.122:3030 ssl:default [Connect call failed ('192.168.50.122', 3030)]
2025-08-03 00:24:52.397 WARNING (MainThread) [custom_components.elegoo_printer] Failed to connect WebSocket to Centauri Carbon: Cannot connect to host 192.168.50.122:3030 ssl:default [Connect call failed ('192.168.50.122', 3030)]
2025-08-03 00:26:15.597 WARNING (MainThread) [custom_components.elegoo_printer] Failed to connect WebSocket to Centauri Carbon: Cannot connect to host 192.168.50.122:3030 ssl:default [Connect call failed ('192.168.50.122', 3030)]
2025-08-03 00:27:39.197 WARNING (MainThread) [custom_components.elegoo_printer] Failed to connect WebSocket to Centauri Carbon: Cannot connect to host 192.168.50.122:3030 ssl:default [Connect call failed ('192.168.50.122', 3030)]
2025-08-03 00:29:02.797 WARNING (MainThread) [custom_components.elegoo_printer] Failed to connect WebSocket to Centauri Carbon: Cannot connect to host 192.168.50.122:3030 ssl:default [Connect call failed ('192.168.50.122', 3030)]
2025-08-03 00:30:26.321 WARNING (MainThread) [custom_components.elegoo_printer] Failed to connect WebSocket to Centauri Carbon: Cannot connect to host 192.168.50.122:3030 ssl:default [Connect call failed ('192.168.50.122', 3030)]
2025-08-03 00:31:49.597 WARNING (MainThread) [custom_components.elegoo_printer] Failed to connect WebSocket to Centauri Carbon: Cannot connect to host 192.168.50.122:3030 ssl:default [Connect call failed ('192.168.50.122', 3030)]
2025-08-03 00:33:12.957 WARNING (MainThread) [custom_components.elegoo_printer] Failed to connect WebSocket to Centauri Carbon: Cannot connect to host 192.168.50.122:3030 ssl:default [Connect call failed ('192.168.50.122', 3030)]
2025-08-03 00:34:36.477 WARNING (MainThread) [custom_components.elegoo_printer] Failed to connect WebSocket to Centauri Carbon: Cannot connect to host 192.168.50.122:3030 ssl:default [Connect call failed ('192.168.50.122', 3030)]
2025-08-03 00:35:59.757 WARNING (MainThread) [custom_components.elegoo_printer] Failed to connect WebSocket to Centauri Carbon: Cannot connect to host 192.168.50.122:3030 ssl:default [Connect call failed ('192.168.50.122', 3030)]
2025-08-03 00:37:23.357 WARNING (MainThread) [custom_components.elegoo_printer] Failed to connect WebSocket to Centauri Carbon: Cannot connect to host 192.168.50.122:3030 ssl:default [Connect call failed ('192.168.50.122', 3030)]
2025-08-03 00:38:46.637 WARNING (MainThread) [custom_components.elegoo_printer] Failed to connect WebSocket to Centauri Carbon: Cannot connect to host 192.168.50.122:3030 ssl:default [Connect call failed ('192.168.50.122', 3030)]
2025-08-03 00:40:10.157 WARNING (MainThread) [custom_components.elegoo_printer] Failed to connect WebSocket to Centauri Carbon: Cannot connect to host 192.168.50.122:3030 ssl:default [Connect call failed ('192.168.50.122', 3030)]
2025-08-03 00:41:33.677 WARNING (MainThread) [custom_components.elegoo_printer] Failed to connect WebSocket to Centauri Carbon: Cannot connect to host 192.168.50.122:3030 ssl:default [Connect call failed ('192.168.50.122', 3030)]
2025-08-03 00:42:56.957 WARNING (MainThread) [custom_components.elegoo_printer] Failed to connect WebSocket to Centauri Carbon: Cannot connect to host 192.168.50.122:3030 ssl:default [Connect call failed ('192.168.50.122', 3030)]
2025-08-03 00:44:20.397 WARNING (MainThread) [custom_components.elegoo_printer] Failed to connect WebSocket to Centauri Carbon: Cannot connect to host 192.168.50.122:3030 ssl:default [Connect call failed ('192.168.50.122', 3030)]
2025-08-03 00:45:43.837 WARNING (MainThread) [custom_components.elegoo_printer] Failed to connect WebSocket to Centauri Carbon: Cannot connect to host 192.168.50.122:3030 ssl:default [Connect call failed ('192.168.50.122', 3030)]
2025-08-03 00:47:07.277 WARNING (MainThread) [custom_components.elegoo_printer] Failed to connect WebSocket to Centauri Carbon: Cannot connect to host 192.168.50.122:3030 ssl:default [Connect call failed ('192.168.50.122', 3030)]
2025-08-03 00:48:30.797 WARNING (MainThread) [custom_components.elegoo_printer] Failed to connect WebSocket to Centauri Carbon: Cannot connect to host 192.168.50.122:3030 ssl:default [Connect call failed ('192.168.50.122', 3030)]
2025-08-03 00:49:54.317 WARNING (MainThread) [custom_components.elegoo_printer] Failed to connect WebSocket to Centauri Carbon: Cannot connect to host 192.168.50.122:3030 ssl:default [Connect call failed ('192.168.50.122', 3030)]
2025-08-03 00:51:17.677 WARNING (MainThread) [custom_components.elegoo_printer] Failed to connect WebSocket to Centauri Carbon: Cannot connect to host 192.168.50.122:3030 ssl:default [Connect call failed ('192.168.50.122', 3030)]
2025-08-03 00:52:41.198 WARNING (MainThread) [custom_components.elegoo_printer] Failed to connect WebSocket to Centauri Carbon: Cannot connect to host 192.168.50.122:3030 ssl:default [Connect call failed ('192.168.50.122', 3030)]
2025-08-03 00:54:04.638 WARNING (MainThread) [custom_components.elegoo_printer] Failed to connect WebSocket to Centauri Carbon: Cannot connect to host 192.168.50.122:3030 ssl:default [Connect call failed ('192.168.50.122', 3030)]
2025-08-03 00:55:28.237 WARNING (MainThread) [custom_components.elegoo_printer] Failed to connect WebSocket to Centauri Carbon: Cannot connect to host 192.168.50.122:3030 ssl:default [Connect call failed ('192.168.50.122', 3030)]
2025-08-03 00:56:51.437 WARNING (MainThread) [custom_components.elegoo_printer] Failed to connect WebSocket to Centauri Carbon: Cannot connect to host 192.168.50.122:3030 ssl:default [Connect call failed ('192.168.50.122', 3030)]
[... cut ...] led ('192.168.50.122', 3030)]
2025-08-03 01:10:45.437 WARNING (MainThread) [custom_components.elegoo_printer] Failed to connect WebSocket to Centauri Carbon: Cannot connect to host 192.168.50.122:3030 ssl:default [Connect call failed ('192.168.50.122', 3030)]
2025-08-03 01:12:08.799 WARNING (MainThread) [custom_components.elegoo_printer] Failed to connect WebSocket to Centauri Carbon: Cannot connect to host 192.168.50.122:3030 ssl:default [Connect call failed ('192.168.50.122', 3030)]
</details>

inventor7777, 2025-08-03:
Also what coderabbit said could be helpful, moving the logs to debug could also work, but others might want these logs. 

[image]

danielcherubini, 2025-08-03:
aha i already did that suggestion

danielcherubini, 2025-08-03:
https://github.com/danielcherubini/elegoo-homeassistant/releases/tag/v1.4.10

Released

inventor7777, 2025-08-03:
@danielcherubini looks great! No more log spam :) 

thanks!

inventor7777, 2025-08-03:
@danielcherubini it's not connecting to my printer now! It has the same IP, I've double checked restarted everything.

inventor7777, 2025-08-03:
Oh lol I hadn't seen that HACS didn't say anything at all even after a restart 😑

danielcherubini, 2025-08-03:
you got the update installed though right?

inventor7777, 2025-08-04:
@danielcherubini do you buy Bambu Lab filament or anything from Bambu? 

I have a MakerWorld account which I post my designs on and I have quite a few points. 

However, I don't do exclusive models so the only thing I can redeem is $40 Bambu Lab gift card, but I don't buy Bambu filament or anything else. I really appreciate your help with these bugs and  for making this integration and I'd really like to give you a EU €40 Bambu gift card if it would be useful for you. If not that's fine but id love to put some of my excess points to good use. 

(I can make a secure pCloud link to the redeem link with a password or whatever for security)

danielcherubini, 2025-08-04:
Wow that's really generous of you. And yes. I do have a bambu A1 and we use a lot of bambulabs filament.

danielcherubini, 2025-08-04:
my email is pretty public, its daniel@cherubini.casa

inventor7777, 2025-08-04:
Okay great I'll send it there instead I didn't even notice that haha

inventor7777, 2025-08-06:
@danielcherubini Bambu finally approved my gift card redemption, I emailed the code to you just now :)

oblivioncth, 2026-10-09:
I don't suppose you'd reconsider a config option to have it so that when the printer is off all sensors simply become Unavailable as originally suggested? I have my printer on only rarely and restart HA far, far more often than the times I turn the printer on, so I pretty much constantly have to look at the integration marked as failed.

While I understand people who leave their printer on all the time would want it that way, and that for many HA integrations this behavior makes sense (and is why the docs recommend marking as failed when a connection error occurs), for devices you don't always have on (especially that you have on rarely), I think it's fair to want a mode where a failed connection should not be viewed as an error state due to it being expected that the device is often off. I understand from a "by the book sense" this may feel weird, but if you look at it practically, it also kind of doesn't make sense that the integration won't be marked as failed or throw tons of errors if the printer is turned on/off a million times after booting, but you're suddenly "punished" for that situation when restarting HA.

A config option to make it so the integration acts just like it would when the printer is later turned off (which it clearly is able to recover from no problem once the printer is turned back on) during setup would be most useful.

I can open a new issue if this should remain as-is.

danielcherubini, 2026-10-09:
Sure that feels like a good compromise that I hadn't considered! 

Thanks for the idea. It was actually really obvious.

oblivioncth, 2026-10-14:
> 18\. Sure that feels like a good compromise that I hadn't considered!
>      Thanks for the idea. It was actually really obvious.

Np, and thanks it works great