danielcherubini/elegoo-homeassistant #203: Cover image transparency

WolfwithSword, 2025-08-17:
### Checklist

- [x] I have filled out the template to the best of my ability.
- [x] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [x] This issue is not a duplicate feature request of [previous feature requests](https://github.com/ludeeus/elegoo_printer/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

(From a Centauri Carbon)

The cover image entity is correct - however each print I've done, it is with a black background with this integration.

The thumbnail itself from the web api on the printer is the same image but as-uploaded from orcaslicer in my case, which generates it with transparency in the background of the model. When I download it manually, it retains this transparency.

### Describe the solution you'd like

I would like if the cover image entity were the one including the transparency, such that I can overlay it as an image inside of a picture-elements card. This would also help in the case if the model's colours were black, so it would not blend in.

### Describe alternatives you've considered

I've tested with a simple websocket connection to fetch the task, get the current thumbnail based off it, and sent in the image to HA over MQTT as an image entity to prove it comes back as transparent in the meantime.

### Additional context

I've done similar via nodered for bambu printers in my HA integration there, and had been working on a similar one for the centauri carbon, but had put my work on pause for several months due to being busy, then stumbled on this integration having almost everything I built already, just a few sensors less and the image issue. Figured I'd switch to it since it's much better put together! Great work here!

danielcherubini, 2025-08-17:
CONFIRMED!

[image]

It is just "whatever is stored"

OK that's kind of easy to solve.

danielcherubini, 2025-08-17:
@WolfwithSword PR is up.

You said something about missing sensors? Which ones?

WolfwithSword, 2025-08-17:
> You said something about missing sensors? Which ones?

It was just a few ones that I was pulling out as basic info before filtering, stuff like X, Y and Z position (doesn't update perfectly in real time, but it's handy to see if you sent movement control commands over at some point)

Diagnostic stuff like the presence of the USB, serial number, storage space remaining, mac address, ip address, current task id etc. Not all of these are probably required, but some are handy if you want to build an automation to pull stuff adhoc with the printer as well (task id, ip, etc).

PR #204 (2025-08-17): FDM Printers use PNGs
Fixes #203 

[image]

[image]

## Summary by CodeRabbit

- **New Features**
  - Thumbnails now preserve/return PNG data for FDM printers and convert non-FDM thumbnails to PNG for consistent handling.
- **Bug Fixes**
  - Thumbnail content type is reported accurately (uses actual image content type).
  - Image bytes, URLs and timestamps are aligned so cached thumbnails display consistently across printer types.