danielcherubini/elegoo-homeassistant #212: Camera feed stops at 60s

bluesk1d, 2025-08-19:
### System Health details

## System Information

version | core-2025.8.2
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
container_arch | amd64
user | root
virtualenv | false
python_version | 3.13.3
os_name | Linux
os_version | 6.12.41-haos
arch | x86_64
timezone | America/Los_Angeles
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
HACS Data | ok
GitHub API Calls Remaining | 5000
Installed Version | 2.0.5
Stage | running
Available Repositories | 2164
Downloaded Repositories | 20

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | July 27, 2026 at 5:00 PM
relayer_connected | true
relayer_region | us-east-1
remote_enabled | true
remote_connected | true
alexa_enabled | false
google_enabled | true
cloud_ice_servers_enabled | true
remote_server | us-east-1-9.ui.nabu.casa
certificate_status | ready
instance_id | 30126d9e79f847b49e9b3249f912ddd6
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 16.1
-- | --
update_channel | stable
supervisor_version | supervisor-2025.08.1
agent_version | 1.7.2
docker_version | 28.3.3
disk_total | 30.8 GB
disk_used | 15.9 GB
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
[... 93 lines cut ...]
2025-08-19 16:18:11.097 DEBUG (MainThread) [custom_components.elegoo_printer] Finished fetching elegoo_printer data in 2.001 seconds (success: True)
2025-08-19 16:18:15.096 DEBUG (MainThread) [custom_components.elegoo_printer.elegoo_sdcp] get_printer_current_task task_id: 7904c0f9-b857-42b6-82c7-76806e99082b
2025-08-19 16:18:15.096 DEBUG (MainThread) [custom_components.elegoo_printer.elegoo_sdcp] get_printer_current_task: got cached task
2025-08-19 16:18:15.096 DEBUG (MainThread) [custom_components.elegoo_printer] Finished fetching elegoo_printer data in 2.002 seconds (success: True)
2025-08-19 16:18:19.097 DEBUG (MainThread) [custom_components.elegoo_printer.elegoo_sdcp] get_printer_current_task task_id: 7904c0f9-b857-42b6-82c7-76806e99082b
2025-08-19 16:18:19.097 DEBUG (MainThread) [custom_components.elegoo_printer.elegoo_sdcp] get_printer_current_task: got cached task
2025-08-19 16:18:19.097 DEBUG (MainThread) [custom_components.elegoo_printer] Finished fetching elegoo_printer data in 2.001 seconds (success: True)
2025-08-19 16:18:23.096 DEBUG (MainThread) [custom_components.elegoo_printer.elegoo_sdcp] get_printer_current_task task_id: 7904c0f9-b857-42b6-82c7-76806e99082b
2025-08-19 16:18:23.096 DEBUG (MainThread) [custom_components.elegoo_printer.elegoo_sdcp] get_printer_current_task: got cached task
2025-08-19 16:18:23.096 DEBUG (MainThread) [custom_components.elegoo_printer] Finished fetching elegoo_printer data in 2.001 seconds (success: True)
2025-08-19 16:18:27.096 DEBUG (MainThread) [custom_components.elegoo_printer.elegoo_sdcp] get_printer_current_task task_id: 7904c0f9-b857-42b6-82c7-76806e99082b
2025-08-19 16:18:27.096 DEBUG (MainThread) [custom_components.elegoo_printer.elegoo_sdcp] get_printer_current_task: got cached task
2025-08-19 16:18:27.096 DEBUG (MainThread) [custom_components.elegoo_printer] Finished fetching elegoo_printer data in 2.001 seconds (success: True)
2025-08-19 16:18:31.096 DEBUG (MainThread) [custom_components.elegoo_printer.elegoo_sdcp] get_printer_current_task task_id: 7904c0f9-b857-42b6-82c7-76806e99082b
2025-08-19 16:18:31.096 DEBUG (MainThread) [custom_components.elegoo_printer.elegoo_sdcp] get_printer_current_task: got cached task
2025-08-19 16:18:31.097 DEBUG (MainThread) [custom_components.elegoo_printer] Finished fetching elegoo_printer data in 2.002 seconds (success: True)
2025-08-19 16:18:32.951 ERROR (Thread-666 (_start_servers_in_thread)) [custom_components.elegoo_printer] Error proxying video stream: 
2025-08-19 16:18:35.096 DEBUG (MainThread) [custom_components.elegoo_printer.elegoo_sdcp] get_printer_current_task task_id: 7904c0f9-b857-42b6-82c7-76806e99082b
2025-08-19 16:18:35.096 DEBUG (MainThread) [custom_components.elegoo_printer.elegoo_sdcp] get_printer_current_task: got cached task
2025-08-19 16:18:35.096 DEBUG (MainThread) [custom_components.elegoo_printer] Finished fetching elegoo_printer data in 2.001 seconds (success: True)
2025-08-19 16:18:39.096 DEBUG (MainThread) [custom_components.elegoo_printer.elegoo_sdcp] get_printer_current_task task_id: 7904c0f9-b857-42b6-82c7-76806e99082b
2025-08-19 16:18:39.096 DEBUG (MainThread) [custom_components.elegoo_printer.elegoo_sdcp] get_printer_current_task: got cached task
2025-08-19 16:18:39.096 DEBUG (MainThread) [custom_components.elegoo_printer] Finished fetching elegoo_printer data in 2.001 seconds (success: True)
2025-08-19 16:21:13.182 ERROR (Thread-666 (_start_servers_in_thread)) [custom_components.elegoo_printer] Error proxying video stream: Cannot write to closing transport
```

danielcherubini, 2025-08-20:
`2025-08-19 16:21:13.182 ERROR (Thread-666 (_start_servers_in_thread)) [custom_components.elegoo_printer] Error proxying video stream: Cannot write to closing transport`

It's this.. i may have solved this in the next version which is in pre-release, if you want to test that out, just go into HACS, and click "redownload" and select the pre-release version.. its at 2.2.1b0 at the moment. 

The issue is two fold 
1. that the web proxy has a 60s timeout. 
2. home assistant's default camera viewer isn't the greatest, i use a different one which doesn't have these issues. I've had similar issues with other cameras.

danielcherubini, 2025-08-20:
https://community.home-assistant.io/t/problem-to-get-cameras-running-constantly/727631/6

FYI

danielcherubini, 2025-08-20:
I believe the new version fixes this!

bluesk1d, 2025-08-21:
Can confirm that 1b1 resolves it, thanks!

I don't use the HA camera entity for anything, just the proxy (unless the video feed being proxied is using HA camera stuff behind the scenes).

danielcherubini, 2025-08-21:
Fixed in 2.2.1