danielcherubini/elegoo-homeassistant #213: Integration won't reconnect to Centauri Carbon after a power loss, leading to have to restart HA

SiDtheTurtle, 2025-08-20:
### System Health details

Running HA core-2025.6.1,

### Checklist

- [x] I have enabled debug logging for my installation.
- [x] I have filled out the issue template to the best of my ability.
- [x] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [x] This issue is not a duplicate issue of any [previous issues](https://github.com/ludeeus/elegoo_printer/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

For safety I have the printer wired up to an IoT plug. Once the last job is done I use the IoT plug to cut the power, and I turn it back on again in the morning when I can queue up another job. The printer works fine, reconnects and I can see its web interface etc.

However, none of the entities in HA update beyond their state at the power loss, but HA is not reporting this as an issue- it thinks everything is okay. Restarting HA fixes the problem.

I suspect the proxy server is key. Even after the power loss the proxy server claims to have successfully connected to the printer, and as the proxy server is running, I think HA doesn't see that there's any issue. More examples in the debug log.

### Reproduction steps

1. Turn off power to the Centauri Carbon at the wall.
2. Turn it back on again.
3. Note none of the entities update.
4. Reboot HA.
5. Entities now update.

### Debug logs

```text
Example log from when the printer was idle, but powered:

2025-08-19 21:42:17.718 INFO (Thread-94 (_start_servers_in_thread)) [custom_components.elegoo_printer] WebSocket client disconnected from 172.19.0.3
2025-08-19 21:42:17.718 INFO (Thread-94 (_start_servers_in_thread)) [custom_components.elegoo_printer] WebSocket client disconnected from 172.19.0.3
2025-08-19 21:42:17.718 INFO (Thread-94 (_start_servers_in_thread)) [custom_components.elegoo_printer] WebSocket client disconnected from 172.19.0.3
2025-08-19 21:42:17.719 INFO (Thread-94 (_start_servers_in_thread)) [custom_components.elegoo_printer] WebSocket client disconnected from 172.19.0.3
2025-08-19 21:42:17.719 INFO (Thread-94 (_start_servers_in_thread)) [custom_components.elegoo_printer] WebSocket client disconnected from 172.19.0.3
2025-08-19 21:42:17.719 INFO (MainThread) [custom_components.elegoo_printer] WebSocket listener stopped.
2025-08-19 21:42:18.754 INFO (MainThread) [custom_components.elegoo_printer] Stopping proxy server...
2025-08-19 21:42:18.758 INFO (MainThread) [custom_components.elegoo_printer] Proxy server stopped.
2025-08-19 21:42:18.759 INFO (MainThread) [custom_components.elegoo_printer] Initializing proxy server for remote printer 172.19.0.3
2025-08-19 21:42:18.764 INFO (Thread-95 (_start_servers_in_thread)) [custom_components.elegoo_printer] Main HTTP/WebSocket Proxy running on http://172.19.0.3:3030
2025-08-19 21:42:18.766 INFO (Thread-95 (_start_servers_in_thread)) [custom_components.elegoo_printer] Video Proxy running on http://172.19.0.3:3031
2025-08-19 21:42:18.768 INFO (Thread-95 (_start_servers_in_thread)) [custom_components.elegoo_printer] Discovery Proxy listening on UDP port 3000
2025-08-19 21:42:18.768 INFO (Thread-95 (_start_servers_in_thread)) [custom_components.elegoo_printer] Unified HTTP/WebSocket Proxy running on http://172.19.0.3:3030
2025-08-19 21:42:18.769 INFO (MainThread) [custom_components.elegoo_printer] Proxy server has started successfully.
2025-08-19 21:42:18.770 DEBUG (MainThread) [custom_components.elegoo_printer] Reconnecting to printer: 172.19.0.3 proxy_enabled True
2025-08-19 21:42:18.770 INFO (MainThread) [custom_components.elegoo_printer] Closing connection to printer
2025-08-19 21:42:18.770 DEBUG (MainThread) [custom_components.elegoo_printer] Connecting to printer: Centauri Carbon at 172.19.0.3 proxy_enabled: True
2025-08-19 21:42:18.771 INFO (MainThread) [custom_components.elegoo_printer] Client connecting to WebSocket at: ws://172.19.0.3:3030/websocket
2025-08-19 21:42:18.777 INFO (MainThread) [custom_components.elegoo_printer] Client successfully connected to: Centauri Carbon, via proxy: True
2025-08-19 21:42:18.777 INFO (MainThread) [custom_components.elegoo_printer] Elegoo printer reconnected successfully.
2025-08-19 21:42:18.777 DEBUG (MainThread) [custom_components.elegoo_printer] Finished fetching elegoo_printer data in 0.023 seconds (success: True)
2025-08-19 21:42:18.782 INFO (Thread-95 (_start_servers_in_thread)) [custom_components.elegoo_printer] WebSocket client connected from 172.19.0.3
2025-08-19 21:42:18.786 INFO (Thread-95 (_start_servers_in_thread)) [custom_components.elegoo_printer] WebSocket client connected from 172.19.0.3
2025-08-19 21:42:18.788 INFO (Thread-95 (_start_servers_in_thread)) [custom_components.elegoo_printer] Proxy connected to remote printer WebSocket at 172.19.0.3
2025-08-19 21:42:18.791 INFO (Thread-95 (_start_servers_in_thread)) [custom_components.elegoo_printer] WebSocket client connected from 172.19.0.3
2025-08-19 21:42:18.793 INFO (Thread-95 (_start_servers_in_thread)) [custom_components.elegoo_printer] Proxy connected to remote printer WebSocket at 172.19.0.3
2025-08-19 21:42:18.796 INFO (Thread-95 (_start_servers_in_thread)) [custom_components.elegoo_printer] WebSocket client connected from 172.19.0.3
2025-08-19 21:42:18.798 INFO (Thread-95 (_start_servers_in_thread)) [custom_components.elegoo_printer] Proxy connected to remote printer WebSocket at 172.19.0.3
2025-08-19 21:42:18.801 INFO (Thread-95 (_start_servers_in_thread)) [custom_components.elegoo_printer] WebSocket client connected from 172.19.0.3
2025-08-19 21:42:18.803 INFO (Thread-95 (_start_servers_in_thread)) [custom_components.elegoo_printer] Proxy connected to remote printer WebSocket at 172.19.0.3

Example after power loss:

2025-08-19 21:52:21.708 INFO (Thread-96 (_start_servers_in_thread)) [custom_components.elegoo_printer] WebSocket client disconnected from 172.19.0.3
2025-08-19 21:52:21.709 INFO (Thread-96 (_start_servers_in_thread)) [custom_components.elegoo_printer] WebSocket client disconnected from 172.19.0.3
2025-08-19 21:52:21.709 INFO (Thread-96 (_start_servers_in_thread)) [custom_components.elegoo_printer] WebSocket client disconnected from 172.19.0.3
2025-08-19 21:52:21.709 INFO (MainThread) [custom_components.elegoo_printer] WebSocket listener stopped.
2025-08-19 21:52:22.754 INFO (MainThread) [custom_components.elegoo_printer] Stopping proxy server...
2025-08-19 21:52:22.757 INFO (MainThread) [custom_components.elegoo_printer] Proxy server stopped.
2025-08-19 21:52:22.759 INFO (MainThread) [custom_components.elegoo_printer] Initializing proxy server for remote printer 172.19.0.3
2025-08-19 21:52:22.764 INFO (Thread-97 (_start_servers_in_thread)) [custom_components.elegoo_printer] Main HTTP/WebSocket Proxy running on http://172.19.0.3:3030
2025-08-19 21:52:22.766 INFO (Thread-97 (_start_servers_in_thread)) [custom_components.elegoo_printer] Video Proxy running on http://172.19.0.3:3031
2025-08-19 21:52:22.767 INFO (Thread-97 (_start_servers_in_thread)) [custom_components.elegoo_printer] Discovery Proxy listening on UDP port 3000
2025-08-19 21:52:22.768 INFO (Thread-97 (_start_servers_in_thread)) [custom_components.elegoo_printer] Unified HTTP/WebSocket Proxy running on http://172.19.0.3:3030
2025-08-19 21:52:22.768 INFO (MainThread) [custom_components.elegoo_printer] Proxy server has started successfully.
2025-08-19 21:52:22.770 DEBUG (MainThread) [custom_components.elegoo_printer] Reconnecting to printer: 172.19.0.3 proxy_enabled True
2025-08-19 21:52:22.770 INFO (MainThread) [custom_components.elegoo_printer] Closing connection to printer
2025-08-19 21:52:22.770 DEBUG (MainThread) [custom_components.elegoo_printer] Connecting to printer: Centauri Carbon at 172.19.0.3 proxy_enabled: True
2025-08-19 21:52:22.770 INFO (MainThread) [custom_components.elegoo_printer] Client connecting to WebSocket at: ws://172.19.0.3:3030/websocket
2025-08-19 21:52:22.777 INFO (MainThread) [custom_components.elegoo_printer] Client successfully connected to: Centauri Carbon, via proxy: True
2025-08-19 21:52:22.777 INFO (MainThread) [custom_components.elegoo_printer] Elegoo printer reconnected successfully.
2025-08-19 21:52:22.777 DEBUG (MainThread) [custom_components.elegoo_printer] Finished fetching elegoo_printer data in 0.023 seconds (success: True)
2025-08-19 21:52:22.782 INFO (Thread-97 (_start_servers_in_thread)) [custom_components.elegoo_printer] WebSocket client connected from 172.19.0.3
[... cut ...]
2025-08-19 21:52:22.791 INFO (Thread-97 (_start_servers_in_thread)) [custom_components.elegoo_printer] Proxy connected to remote printer WebSocket at 172.19.0.3
2025-08-19 21:52:22.792 INFO (Thread-97 (_start_servers_in_thread)) [custom_components.elegoo_printer] WebSocket client connected from 172.19.0.3

It seems to be exactly the same, claiming it's connected when it's not. This pattern in the logs continues all night, even though the printer is off, and continues now it's back on again, but in reality HA is not updating.

Note the IP range is not that of my local network, I assume that is something the proxy server is doing.
```

danielcherubini, 2025-08-20:
Oh that im aware of :D 

Using the smart plug just enables me to do what i already do with my Elegoo Resin printer .. restarting it without getting off my ass

danielcherubini, 2025-08-20:
https://github.com/danielcherubini/elegoo-homeassistant/releases/tag/v2.2.1b1
OK Pre-release is up for testing

SiDtheTurtle, 2025-08-21:
Confirmed fixed with v2.2.1, powdered down, waited and while and powered up, HA gracefully noticed and reconnected to the device once it was back up. Cheers for the quick turnaround Daniel!

danielcherubini, 2025-08-21:
You're welcome :D