danielcherubini/elegoo-homeassistant #230: Proxy Ports are already in use

ChristophCaina, 2025-08-25:
### System Health details

None

### Checklist

- [x] I have enabled debug logging for my installation.
- [x] I have filled out the issue template to the best of my ability.
- [x] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [x] This issue is not a duplicate issue of any [previous issues](https://github.com/ludeeus/elegoo_printer/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

I noticed that my printer became unavailable from time to time, but only in homeassistant. 
In chitu Manager, the printer was still available.

Today, the printer became unavailable for longer then usually, so I tried to restart the Integration. 

Now, I am getting this Error:
Proxy Ports are already in use. 

I beleive, that there is an issue with the stability of the WiFi to the printer, because from time to time I can also See the printer being shown offline in chitu Manager - but right now, I am Not Sure If this might be an issue from the Integration

### Reproduction steps

1.
2.
3.
...

### Debug logs

```text
None
```

ChristophCaina, 2025-08-25:
Forgot to add a screenshot[image]

[image]

danielcherubini, 2025-08-25:
Ok. For starters. You have a resin printer. Just disable the proxy. You don't need it. 

It's for the Centauri carbon. I didn't have good documentation around this when you installed it. But since then it's been removed from the options when adding a new printer. I have kept it in the options on an existing printer. 

So yeah. Long term. Disable it. 

Short term. This is a temp problem. I've got a pretty hefty rewrite of the proxy for multiple printers coming. It resolves this issue specially.

ChristophCaina, 2025-08-25:
Thanks :)
At the time of writing the Ticket, i could Not disable the Proxy.
It Just didn't Update the setting.

I noticed now, that my repeater was also kicked Out of the Mesh Network and could No longer be reached, but the Last days, the printer did Connect to a different repeater.
The printer did also Show, that it could Not connect to the WiFi for some reasons.

Seems, that Something with my whole Network was way Off..
I will probably investigate Intro this Fürther tomorrow - but thanks for the Tip with the Proxy... I will disable it at all

danielcherubini, 2025-08-25:
> At the time of writing the Ticket, i could Not disable the Proxy.
It Just didn't Update the setting.

easy solution is to remove the printer and re-add it you don't lose anything as you dont delete history .. and re-adding just puts it back and HA connects the history.. (theres a unique ID which is your printer serial number)

ChristophCaina, 2025-08-26:
yeah, I got it resolved yesterday night...
Still, the camera stream breaks up from time to time, but right now, it is only showing one connection at a time (I think, the stream is still struggling with the wifi quality in that area)...

This topic can be closed as of now - since the issue was a local network issue and restarting the printer did solve it for now.