danielcherubini/elegoo-homeassistant #235: Remaining Print Time/End Time aren't accurate

inventor7777, 2025-08-30:
### System Health details

Same as issues I've submitted before

I don't use the proxy server btw

### Checklist

- [x] I have enabled debug logging for my installation.
- [x] I have filled out the issue template to the best of my ability.
- [x] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [x] This issue is not a duplicate issue of any [previous issues](https://github.com/ludeeus/elegoo_printer/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

Sometimes on long prints when the print finishes the print time/end time sensors don't update properly.

[image]

I know this could be an Elegoo issue, but I'm just putting it here just in case

### Reproduction steps

1. Start super long print
2. wait for print to finish
3. Check end time/remaining print time sensors

### Debug logs

```text
2025-08-30 17:07:45.496 DEBUG (MainThread) [custom_components.elegoo_printer] Finished fetching Centauri Carbon data in 0.322 seconds (success: True)
2025-08-30 17:07:47.505 DEBUG (MainThread) [custom_components.elegoo_printer] Finished fetching Centauri Carbon data in 0.333 seconds (success: True)
2025-08-30 17:07:49.505 DEBUG (MainThread) [custom_components.elegoo_printer] Finished fetching Centauri Carbon data in 0.332 seconds (success: True)
2025-08-30 17:07:51.517 DEBUG (MainThread) [custom_components.elegoo_printer] Finished fetching Centauri Carbon data in 0.343 seconds (success: True)
2025-08-30 17:07:53.543 DEBUG (MainThread) [custom_components.elegoo_printer] Finished fetching Centauri Carbon data in 0.370 seconds (success: True)
2025-08-30 17:07:55.567 DEBUG (MainThread) [custom_components.elegoo_printer] Finished fetching Centauri Carbon data in 0.395 seconds (success: True)
2025-08-30 17:07:57.575 DEBUG (MainThread) [custom_components.elegoo_printer] Finished fetching Centauri Carbon data in 0.402 seconds (success: True)
2025-08-30 17:07:59.607 DEBUG (MainThread) [custom_components.elegoo_printer] Finished fetching Centauri Carbon data in 0.434 seconds (success: True)
2025-08-30 17:08:01.526 DEBUG (MainThread) [custom_components.elegoo_printer] Finished fetching Centauri Carbon data in 0.354 seconds (success: True)
2025-08-30 17:08:03.547 DEBUG (MainThread) [custom_components.elegoo_printer] Finished fetching Centauri Carbon data in 0.373 seconds (success: True)
2025-08-30 17:08:05.570 DEBUG (MainThread) [custom_components.elegoo_printer] Finished fetching Centauri Carbon data in 0.397 seconds (success: True)
2025-08-30 17:08:07.596 DEBUG (MainThread) [custom_components.elegoo_printer] Finished fetching Centauri Carbon data in 0.423 seconds (success: True)
2025-08-30 17:08:09.498 DEBUG (MainThread) [custom_components.elegoo_printer] Finished fetching Centauri Carbon data in 0.325 seconds (success: True)
2025-08-30 17:08:11.496 DEBUG (MainThread) [custom_components.elegoo_printer] Finished fetching Centauri Carbon data in 0.322 seconds (success: True)
2025-08-30 17:08:13.498 DEBUG (MainThread) [custom_components.elegoo_printer] Finished fetching Centauri Carbon data in 0.326 seconds (success: True)
2025-08-30 17:08:13.902 INFO (MainThread) [custom_components.elegoo_printer] Closing connection to printer
2025-08-30 17:08:13.904 ERROR (MainThread) [homeassistant.components.template.template_entity] TemplateError('ValueError: Template error: float got invalid input 'unavailable' when rendering template '{% set remaining = states('sensor.centauri_carbon_remaining_print_time') | float %}
{% set hours = remaining | int %}
{% set minutes = ((remaining - hours) * 60) | round(0) | int %}

{% set end_ts = as_timestamp(states('sensor.centauri_carbon_end_time')) %}
{% set end_time = end_ts | timestamp_custom('%H:%M') %}

{{ hours }}h {{ minutes }}m / {{ end_time }}' but no default was specified') while processing template 'Template<template=({% set remaining = states('sensor.centauri_carbon_remaining_print_time') | float %}
{% set hours = remaining | int %}
{% set minutes = ((remaining - hours) * 60) | round(0) | int %}

{% set end_ts = as_timestamp(states('sensor.centauri_carbon_end_time')) %}
{% set end_time = end_ts | timestamp_custom('%H:%M') %}

{{ hours }}h {{ minutes }}m / {{ end_time }}) renders=47406>' for attribute '_attr_native_value' in entity 'sensor.centauri_carbon_time_info'
2025-08-30 17:08:13.906 ERROR (MainThread) [homeassistant.components.template.template_entity] TemplateError('ValueError: Template error: as_timestamp got invalid input 'unavailable' when rendering template '{{ as_timestamp(states('sensor.centauri_carbon_end_time')) | timestamp_custom('%H:%M') }}' but no default was specified') while processing template 'Template<template=({{ as_timestamp(states('sensor.centauri_carbon_end_time')) | timestamp_custom('%H:%M') }}) renders=6>' for attribute '_attr_native_value' in entity 'sensor.centauri_carbon_end_time_plus'
2025-08-30 17:08:13.910 DEBUG (MainThread) [custom_components.elegoo_printer] WebSocket listener cancelled.
2025-08-30 17:08:13.911 INFO (MainThread) [custom_components.elegoo_printer] WebSocket listener stopped.
2025-08-30 17:08:13.954 DEBUG (MainThread) [custom_components.elegoo_printer] CONFIGURATION {'brand': 'ELEGOO', 'connection': '979d4C788A4a78bC777A870F1A02867A', 'firmware': 'V1.1.29', 'id': '0c6817140107202800007c0000000000', 'ip_address': '192.168.50.122', 'model': 'Centauri Carbon', 'name': 'Centauri Carbon', 'printer_type': 'fdm', 'protocol': 'V3.0.0', 'proxy_enabled': False}
2025-08-30 17:08:13.954 DEBUG (MainThread) [custom_components.elegoo_printer] Connecting to printer: Centauri Carbon at 192.168.50.122 with proxy enabled False
2025-08-30 17:08:13.955 INFO (MainThread) [custom_components.elegoo_printer] Closing connection to printer
2025-08-30 17:08:13.958 INFO (MainThread) [custom_components.elegoo_printer] Connecting to printer: Centauri Carbon at 192.168.50.122 proxy_enabled: False
2025-08-30 17:08:14.000 ERROR (MainThread) [homeassistant.components.glances.coordinator] Error fetching glances - raspberrypihole.local data: 
2025-08-30 17:08:14.017 INFO (MainThread) [custom_components.elegoo_printer] Client successfully connected to: Centauri Carbon, via proxy: False
2025-08-30 17:08:14.018 INFO (MainThread) [custom_components.elegoo_printer] Polling Started
2025-08-30 17:08:14.406 DEBUG (MainThread) [custom_components.elegoo_printer] Finished fetching Centauri Carbon data in 0.388 seconds (success: True)
2025-08-30 17:08:14.407 DEBUG (MainThread) [custom_components.elegoo_printer] Adding 33 sensor entities
2025-08-30 17:08:14.474 DEBUG (MainThread) [custom_components.elegoo_printer] Adding 2 binary sensor entities
2025-08-30 17:08:14.479 DEBUG (MainThread) [custom_components.elegoo_printer] Adding 1 image entities
2025-08-30 17:08:14.482 DEBUG (MainThread) [custom_components.elegoo_printer] Adding 1 Camera entities
2025-08-30 17:08:14.485 DEBUG (MainThread) [custom_components.elegoo_printer] Adding 1 light entities
2025-08-30 17:08:14.493 DEBUG (MainThread) [custom_components.elegoo_printer] Adding 3 button entities
2025-08-30 17:08:14.819 DEBUG (MainThread) [custom_components.elegoo_printer] Finished fetching Centauri Carbon data in 0.411 seconds (success: True)
2025-08-30 17:08:16.158 DEBUG (MainThread) [custom_components.elegoo_printer] Sending printer video: {'status': <ElegooVideoStatus.SUCCESS: 0>, 'video_url': '192.168.50.122:3031/video'}
2025-08-30 17:08:16.158 DEBUG (MainThread) [custom_components.elegoo_printer] stream_source: Video is OK, getting stream source
2025-08-30 17:08:16.158 DEBUG (MainThread) [custom_components.elegoo_printer] stream_source: Proxy is disabled using printer video url: 192.168.50.122:3031/video
2025-08-30 17:08:17.555 DEBUG (MainThread) [custom_components.elegoo_printer] Finished fetching Centauri Carbon data in 0.343 seconds (success: True)
2025-08-30 17:08:19.576 DEBUG (MainThread) [custom_components.elegoo_printer] Finished fetching Centauri Carbon data in 0.364 seconds (success: True)
2025-08-30 17:08:20.110 DEBUG (MainThread) [custom_components.elegoo_printer] Sending printer video: {'status': <ElegooVideoStatus.SUCCESS: 0>, 'video_url': '192.168.50.122:3031/video'}
2025-08-30 17:08:20.111 DEBUG (MainThread) [custom_components.elegoo_printer] stream_source: Video is OK, getting stream source
2025-08-30 17:08:20.111 DEBUG (MainThread) [custom_components.elegoo_printer] stream_source: Proxy is disabled using printer video url: 192.168.50.122:3031/video
[... cut ...]
2025-08-30 17:08:25.569 DEBUG (MainThread) [custom_components.elegoo_printer] Finished fetching Centauri Carbon data in 0.357 seconds (success: True)
2025-08-30 17:08:27.582 DEBUG (MainThread) [custom_components.elegoo_printer] Finished fetching Centauri Carbon data in 0.369 seconds (success: True)
2025-08-30 17:08:29.596 DEBUG (MainThread) [custom_components.elegoo_printer] Finished fetching Centauri Carbon data in 0.384 seconds (success: True)
2025-08-30 17:08:31.605 DEBUG (MainThread) [custom_components.elegoo_printer] Finished fetching Centauri Carbon data in 0.394 seconds (success: True)
```

danielcherubini, 2025-08-30:
This isn't a bug with the integration.. it's a printer issue. Trust me I noticed this too and I did extensive debugging. I'm showing you the exact numbers it gets from the printer. 

Its only until recently with orca did we even get the correct number of layers being counted. 

My guess is this gets fixed in a firmware update one day.

danielcherubini, 2025-08-30:
That said. I noticed in your debug logs a problem with the notification blueprint! 

Gonna leave this open and fix that.

inventor7777, 2025-08-30:
@danielcherubini thats probably not the blueprint, that's a custom template sensor I made that shows remaining time / end time

Example: 4hr5m / 16:30

closed 2025-08-30 by inventor7777

danielcherubini, 2025-08-31:
ahhhhhh! 

That said, the ONLY real thing i can do here, is set the remaining_time and end_time to 0 or there abouts when the print has completed normally. But that comes at the cost of fixing an issue that's not really my issue to fix. 

So the only time i did this so far was for the percentage, before the printer would send the percent complete as it ticks up to 100% then when it gets to 100% the print would complete, and the printer would now send 0%

So 100% would happen for a split second, which kind of sucks if you're writing automations against that, and also would look real weird in the dashboard when the print says complete and the percentage is 0%.

I do however have some faith that this issue should get patched soon. 

For reference, here's how Orca solved the layer issue
https://github.com/SoftFever/OrcaSlicer/pull/9630

They basically created a macro that runs on each layer, incrementing it.. something the printer should do.

inventor7777, 2025-08-31:
I'm happy with it the way it is. Since it's a printer issue I doubt it's worth trying to fix it here, especially as Elegoo will hopefully update it eventually :)