danielcherubini/elegoo-homeassistant #253: notify on print errors and not just when complete

mmerickel, 2025-09-26:
### Checklist

- [x] I have filled out the template to the best of my ability.
- [x] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [x] This issue is not a duplicate feature request of [previous feature requests](https://github.com/ludeeus/elegoo_printer/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

The issue that's causing me to bring this up is when I ran into a filament runout. The print is now paused but there's no update to the notifications so I check my progress notification and think things are ok other than that the progress hasn't updated in a while. Then I check the CC UI and it's out of filament :-(

### Describe the solution you'd like

The current blueprint only tracks print progress and then a final complete notification. There are no notifications for other sub-states of printing. 

The supported "print_status" contains the following:

```
  - idle
  - homing
  - dropping
  - printing
  - lifting
  - pausing
  - paused
  - stopping
  - stopped
  - complete
  - file_checking
  - recovery
  - printing_recovery
  - loading
```

I think it'd be useful to expose the other states while the printer's top-level "current_status" (not "print_status" to be clear) is "printing". While in that mode it'd be nice to just have the notification always updated. Meaning it'd show a notification for every state.

- Notification for progress while actively printing.
- Notification for successful completion.
- Notification showing current state when not actively progressing (`print_status not in ["printing", "complete"]`). For example "Print status: homing".

### Describe alternatives you've considered

I can obviously DIY my own notification, this request is just to see if it's interesting to include in the builtin blueprint.

### Additional context

loving the addon!

danielcherubini, 2025-09-26:
Yeah totally agree. I'll get on that today. 

I want to mention two things. 

1. Blueprints need to be manually updated. But I'll create a release in GitHub which should create a notification to update for everyone on their HA.. ill mention in the release notes to manually update the blueprint. (It's annoying I know)
2. I have actually added a loading/unloading filament status too. So that will be picked up. 

Also with filament jams, there's a status for that.. the current toolhead doesn't have a sensor for that.. but it's actually in the new firmware, which indicates either a new toolhead for the printer with more sensors, or something we add too the toolhead with those sensors. So with hopes... This blueprint will just pick those up.

danielcherubini, 2025-09-27:
OK its there .. check the release notes.